gdsfactory/gdsfactory #15: Geometry import

smartalecH, 2020-10-30:
Rather than export to an HDF5 file...
https://github.com/gdsfactory/gdsfactory/blob/f47be487cc84a5e6c727d4326da6292a0f15d981/gdsmp/sim.py#L18

I would instead directly import the geometry using prism objects. Since you already have the polygon information from the gds "formalism", it's easy to import those.

This allows you to easily use dispersive materials, doesn't involve an extra layer of interpolation, and can even leverage objects with sidewall angles.

smartalecH, 2020-10-30:
Ah I see. The vertices are exported to an hdf5 file. This step could still probably be skipped -- the vertices can be imported directly using gdspy.

joamatab, 2020-11-02:
yes Alec, I didn't get too far with that file. The best one is simulate_sp.py which is based on your taper example

i would expect the Sparameters for a straight waveguide to be 0dB transmission ^^

not sure what's going on

smartalecH, 2020-11-02:
There are lot's of issues with the script, the first being the geometry isn't a straight waveguide:

[image]

smartalecH, 2020-11-02:
The two biggest issues are the placement of your monitors and calculation of the s parameters. Your monitors should be placed wherever you want to define the device edges. The corresponding source need to be defined outside of the monitors:

[image]

You need to calculate the corresponding s params like so:
```python
        # Parse out the overlaps
        a1 = m1_results[:, :, 0]  # forward wave
        b1 = m1_results[:, :, 1]  # backward wave
        a2 = m2_results[:, :, 0]  # forward wave
        b2 = m2_results[:, :, 1]  # backward wave

        # Calculate the actual scattering parameters from the overlaps
        S11 = np.squeeze(b1 / a1)
        S12 = np.squeeze(a2 / a1)
```
[image]

The geometry error that you see above is a discretization error from the prism objects being right next to each other. The error approaches zero rather quickly as resolution increases.

I can submit a PR with the whole script.

joamatab, 2020-12-20:
Thank you Alec

PR #18 (2020-11-03): Fix S param script
Closes #15.

Note I'm merging into the "meep" branch, not master.
closed 2020-12-20 by joamatab