gdsfactory/gdsfactory #53: disconnected waveguides in some manhattan routes

joamatab, 2021-05-28:
How can we fix some disconnected waveguides in some manhattan routes?

```
if __name__ == "__main__": 
    import pp
    c = pp.Component()
    c1 = c << pp.c.straight_array(spacing=50)
    c2 = c << pp.c.straight_array(spacing=5)
    c2.movex(200)
    c1.y = 0
    c2.y = 0

    routes = pp.routing.get_bundle_path_length_match(
        c1.get_ports_list(orientation=0),
        c2.get_ports_list(orientation=180),
        end_straight_offset=0,
        start_straight=0,
        separation=50,
    )

    for route in routes:
        c.add(route.references)
    c.show()
```

[image]

joamatab, 2022-08-20:
We can extend the new Sven's option to fix the caes with impossible manhattan routes

SkandanC, 2022-09-14:
@joamatab 

[image]
I added an bit of code to return s_bend bundle routes if a RouteWarning is raised and it gives me this.

What do you think?

joamatab, 2022-09-14:
For path length matching sbends have different length so it may not be a good solution

Hezq23, 2022-12-23:
I have encountered the same problem when I'm using the gdsfactory, I just mitigate it by using the partially bent waveguides to route instead of 90' waveguide bend. (I have written the routing function myself.) I know in gdsfactory, the waveguide bend used in route can only be 90', this is unnecessary and will sometimes cause trouble like the issue discussed.

joamatab, 2022-12-23:
Yes, when this happens I usually change the position of one of the components that we are routing to

joamatab, 2023-10-13:
here is the updated code for the issue 

```python
"""
Some manhattan routes have disconnected waveguides

"""

import gdsfactory as gf

if __name__ == "__main__":
    c = gf.Component()
    # c1 = c << gf.components.straight_array(spacing=200)
    c1 = c << gf.components.straight_array(spacing=50)
    c2 = c << gf.components.straight_array(spacing=5)

    c2.movex(200)
    c1.y = 0
    c2.y = 0

    routes = gf.routing.get_bundle_path_length_match(
        c1.get_ports_list(orientation=0),
        c2.get_ports_list(orientation=180),
        end_straight_length=0,
        start_straight_length=0,
        separation=50,
        radius=10,
        # radius=3 # smaller radius works
        enforce_port_ordering=False
    )

    for route in routes:
        c.add(route.references)

    c.show()
```

closed 2023-12-21 by github-actions[bot]