gdsfactory/gdsfactory #56: fix mypy issues

joamatab, 2021-05-30:
Having clearly defined type annotations helps catch many errors on the code

it would be great having `make mypy` give no type errors.

currently we have > 900 errors

elamdf, 2023-03-07:
Working on this over [here](https://github.com/elamdf/gdsfactory/tree/fix_mypy), I think it will be a nontrivial undertaking so I may try to take it in segments.

sebastian-goeldi, 2023-03-08:
I have a few tips (feel free to ignore if you feel comfortable with mypy etc already)

- I strongly suggest to use a plugin that can run in your editor, otherwise you will have to navigate thousands of errors and warnings. I had good experiences with `pyls-mypy`. I believe Joaquin copied settings it over when switching to `pyproject.toml`, so it should be configured correctly automatically.
- strict mode can be annoying, because some times it can feel very much feel like "99 bugs in my code... Fix one, fix another... 154 bugs in my code...". Don't ne discouraged by this ;)
- feel free to tag me if you find a weird one and would like some help. I think I have seen most of the mypy errors you will encounter

closed 2023-11-11 by github-actions[bot]