gdsfactory/gdsfactory #57: implement route optimizer

joamatab, 2021-06-14:
It would be great to implement a router optimizer where you can penalize different things:

- minimize number of waveguide crossings
- keep path length match between groups of routes

joamatab, 2022-08-24:
This looks promising

https://github.com/zhm-real/PathPlanning

the router from openRoad also looks great

https://github.com/The-OpenROAD-Project/OpenLane

joamatab, 2022-09-02:
ideas to get started:

- single route
- bundles of routes
- different bundles competing for space

@tvt173 
@SkandanC 
@thomasdorch

SkandanC, 2022-09-05:
@joamatab Are we planning on adding this as an extension to Manhattan routing, etc. or as its own separate thing?

joamatab, 2022-09-05:
I think we need a separate function for this

joamatab, 2022-09-10:
for routing a group of routes, the following routers seem more promising

- https://codeberg.org/LibrEDA/mycelium-router/src/branch/main/src
- https://codeberg.org/tok/librecell/src/branch/master/librecell-layout/lclayout/graphrouter
- https://github.com/The-OpenROAD-Project/TritonRoute/
- https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/grt/README.md

joamatab, 2022-12-23:
Troy, 

I think you were going to send a PR with a non-Manhattan router

@Hsuan-Tung
@nikosavola 
@95-martin-orion

joamatab, 2023-04-07:
https://github.com/instadeepai/jumanji

SkandanC, 2023-04-08:
KFactory has a draft auto router writte using jumanji. See here:
https://github.com/gdsfactory/kfactory/blob/auto-router/src/kfactory/routing/auto_router.py

@joamatab

SkandanC, 2023-04-09:
I've added a draft auto-router for gdsfactory in the auto-router branch. If you want you can take a look, if anyone would like to collborate and work out how we can get this working better, that'd be great :)

95-martin-orion, 2023-11-03:
Bumping this issue to brush off the dust:
- #2259 was motivated by issues using `get_bundle_all_angle` with manually-defined steps
- Other auto-routing tools would also benefit from better documentation. My team reports that wiring constitutes a significant fraction of their time spent laying out a device; if auto-routing works well, it could greatly reduce that time (and we should advertise that to users)

joamatab, 2023-11-03:
yes, I looked at some PCB routing packages

- https://github.com/freerouting/freerouting/issues/132#issuecomment-1210868519  (java) 

as well as CMOS routers
- https://github.com/The-OpenROAD-Project/TritonRoute/   (C++)
- https://codeberg.org/LibrEDA/mycelium-router  (rust)

Skandan had a demo with 
- https://github.com/instadeepai/jumanji

PRs are welcome

@basnijholt

juanferv, 2024-05-16:
Just in case it helps development, we have tried jumaji's auto-router, but it takes a few minutes to do many iterations. This code is scanning the networkX library to do the grid and pathfinding. Actually, with a grid 5 times denser than the Astar example included in gdsfactory, it takes a couple of seconds compared to several minutes with the get_rout_astar function.
Problems I have detected with the code:
- it fails me when I try to put bends in the routing (I am not an expert and it will be some waypoints nonsense).
- the issue of the orientation of the ports, it would be necessary to “correct” at the beginning or at the end so that “it doesn't matter the orientation” but that it connects well. For metal routing, it is ok, but for optical routing it can be a problem.
- To make multiport, with networkX from what I have seen is quite possible, and handles large grids with ease, so multiport, could be very interesting.
- We have put some loops that search the whole grid to make distance calculations that maybe could be changed by direct algorithms and speed up.
- It occurred to us that we could launch this type of automatic routing in parallel using multithreading or multicore.

from __future__ import annotations

import warnings

import numpy as np

import gdsfactory as gf
from gdsfactory import Port
from gdsfactory.component import Component
from gdsfactory.typings import CrossSectionSpec, LayerSpec, Route
import networkx as nx
from shapely.geometry import LineString

class Node:
    def __init__(self, parent=None, position: tuple = ()) -> None:
        """Initializes a node. A node is a point on the grid."""
        self.parent = parent  # parent node of current node
        self.position = position  # position of current node

        self.g = 0  # distance between current node and start node
        self.h = 0  # distance between current node and end node
        self.f = self.g + self.h  # cost of the node (sum of g and h)

def _extract_all_bbox(c: Component, avoid_layers: list[LayerSpec] | None = None):
    """Extract all polygons whose layer is in `avoid_layers`."""
    return [c.get_polygons(layer) for layer in avoid_layers]

def _generate_grid(
    c: Component,
    resolution: float = 0.5,
    avoid_layers: list[LayerSpec] | None = None,
    distance: float = 1,
) -> np.ndarray:
    """Generate discretization grid that the algorithm will step through."""
    bbox = c.bbox
    x, y = np.meshgrid(
        np.linspace(
            bbox[0][0]-resolution,
            bbox[1][0]+resolution,
            int((bbox[1][0] - bbox[0][0]+2*resolution) / resolution),
            endpoint=True,
        ),
        np.linspace(
            bbox[0][1]-resolution,
            bbox[1][1]+resolution,
            int((bbox[1][1] - bbox[0][1]+2*resolution) / resolution),
            endpoint=True,
        ),
    )  # discretize component space
    x, y = x[0], y[:, 0]  # weed out copies
    grid = np.zeros(
        (len(x), len(y))
    )  # mapping from gdsfactory's x-, y- coordinate to grid vertex

    # assign 1 for obstacles
    if avoid_layers is None:
        for ref in c.references:
            bbox = ref.bbox
            xmin = np.abs(x - bbox[0][0] + distance).argmin()
            xmax = np.abs(x - bbox[1][0] - distance).argmin()
            ymin = np.abs(y - bbox[0][1] + distance).argmin()
            ymax = np.abs(y - bbox[1][1] - distance).argmin()
[... 183 lines cut ...]
obstacle1.xmin = -10
obstacle2.xmin = 35
obstacle3.ymin = 42
obstacle3.xmin = 72.23
obstacle4.xmin = 200
obstacle4.ymin = 55
obstacle5.xmin = 600
obstacle5.ymin = 200
port1 = left.ports["e2"]
port2 = right.ports["e2"]

routes = get_route_networkx2(
    component=c,
    port1=port1,
    port2=port2,
    cross_section=cross_section,
    resolution=1,
    distance=12,
    avoid_layers=("M2",),
)

# for route in routes:
#     c.add(route.references)
c.add(routes.references)
c.show()

joamatab, 2024-05-16:
Thank you for sharing Juan, this works great for metal

i was able to make some cases with optical routes using your code

[image]

the trick is to have radius smaller than the resolution of the grid

sebastian-goeldi, 2026-09-14:
This by now is super old. Something like this we have already in gf+

closed 2026-09-14 by sebastian-goeldi