gdsfactory/gdsfactory #77: Regarding/importing a GDS file to gdsfactory and creating array

ratul619, 2021-11-13:
Hi , 

I want to import an already created GDS file , and then create an array of the shapes that are inside the GDS.
Please let me know the relevant sections in the tutorial i should refer to. I was not able to locate the part of "import an already created GDS file" to gdsfactory. 

Thanks

joamatab, 2021-12-28:
Hi Ratul,

i made a tutorial section specific for showing how to import [GDS files](https://gdsfactory.readthedocs.io/en/latest/notebooks/05_GDS.html)

the array part was already quite well documented in the [references tutorial](https://gdsfactory.readthedocs.io/en/latest/notebooks/01_references.html)

Hope this helps
Joaquin

closed 2021-12-28 by joamatab

ratul619, 2022-02-10:
The link seems broken !

joamatab, 2022-02-10:
yes, i moved the tutorial from readthedocs to github pages

https://gdsfactory.github.io/gdsfactory/

njcoburn, 2023-03-30:
Hey @joamatab, can you point me to where the example is exactly? I cannot seem to find it.
Like the above, I have several different .gds files already made. I would like to pull them into gdsfactory, specifically to employ the automated routing through yaml scripts. Like the attached image.
Many thanks for your help[image]

njcoburn, 2023-03-30:
Just a follow up... I believe that the answer is here: https://gdsfactory.github.io/gdsfactory/_autosummary/gdsfactory.read.import_gds.html?highlight=gf+import_gds