gdsfactory/gdsfactory #133: write_sparameters_meep working with multiple threads

joamatab, 2022-01-08:
one way to make this work is to write a script to disk and then Popen a process

Similar to what Derek does in picwriter
https://github.com/DerekK88/PICwriter/blob/master/picwriter/picsim.py

related to #124  and also suggested by @smartalecH  and @simbilod

Let me know what you think
Joaquin

smartalecH, 2022-01-08:
I could be wrong, but I don't think Popen lets you spawn multiple MPI processes. 

Also note the difference between multi threading and multi-process (meep uses the latter).

joamatab, 2022-01-08:
great,

he actually uses `call` instead of `Popen`, which makes the call blocking

https://stackoverflow.com/questions/7681715/whats-the-difference-between-subprocess-popen-and-call-how-can-i-use-them

simbilod, 2022-01-17:
See PR#149

Instead of using a special simulation file with hardcoded command line arguments as is done in the picwriter example, I write temporary Python scripts that are executed using mpi. This is much better to handle arbitrary **kwargs.

To do this I need to pickle the component object to disk so that I can load it in the generated python file. @joamatab can this cause any problem?

This implementation also goes a little beyond what was requested by allowing multiple different simulations to run in pools. This is useful for parameter sweeps to avoid diminishing returns of large core numbers on single simulations. The user can always just provide a single instance to simulate in the list, and dedicate all cores to it.

I didn't know what test to add for this, suggestions welcome.

## Example

Define 3 different simulation instances (dictionaries):

```
# Some component
c1 = gf.c.straight(length=5)
p = 3
c1 = gf.add_padding_container(c1, default=0, top=p, bottom=p)

# Another component
c2 = gf.c.straight(length=4)
p = 3
c2 = gf.add_padding_container(c2, default=0, top=p, bottom=p)

# First simulation of first component
c1_dict = {
    "component": c1,
    "run": True,
    "overwrite": True,
    "lazy_parallelism": True,
    "filepath":"c1_dict.csv"
}
# Second simulation: second component
c2_dict = {
    "component": c2,
    "run": True,
    "overwrite": True,
    "lazy_parallelism": True,
    "filepath":"c2_dict.csv"
}
# Third simulation: second component, but other parameters different
c3_dict = {
    "component": c2,
    "run": True,
    "overwrite": True,
    "lazy_parallelism": True,
    "resolution": 40,
    "port_source_offset":0.3,
    "filepath":"c3_dict.csv"
}
```
Bundle the instances, and pass it to the parallel function:
```
# Instances
instances = [
    c1_dict,
    c2_dict,
    c3_dict,
]

write_sparameters_meep_parallel(
        instances=instances,
        cores_per_instance=4,
        total_cores=10,
        delete_temp_files=False,
        verbosity=True,
    )
```
I want to use a total of 10 cores, and 4 cores per instance. So the code will run the first 2 together, and the third one afterwards.

The output is as follows (with verbosity True, there is some extra output):

```
Running parallel simulations over 3 instances
Using a total of 10 cores with 4 cores per instance
Tasks split amongst 2 pools with up to 2 instances each.
Task 0 of pool 0 is instance 0
Task 1 of pool 0 is instance 1
Launching pool 0 simulations, which contain commands:
mpirun -np 4 python /home/simbil/Github/gdsfactory/gdslib/sp/temp/write_sparameters_meep_parallel_0.py
mpirun -np 4 python /home/simbil/Github/gdsfactory/gdslib/sp/temp/write_sparameters_meep_parallel_1.p
```
At this point, two 4-core MPI simulations are properly running:
[image]

When these are done, we continue to the next pool:

```
Task 0 of pool 1 is instance 2
Launching pool 1 simulations, which contain commands:
mpirun -np 4 python /home/simbil/Github/gdsfactory/gdslib/sp/temp/write_sparameters_meep_parallel_2.py
```
As expected, only 4 cores are used now:
[image]

The outputs are saved where we want:
[image]

We can also inspect the temporary Python scripts (there is a flag to automatically delete them):

[image]

simbilod, 2022-01-17:
I also added a version of this without the pooling:

```
def write_sparameters_meep_parallel(
    instance: Dict,
    cores: int = 2,
    temp_dir: Path = CONFIG["sparameters"] / "temp",
    delete_temp_files: bool = True,
)
```

In this case, it returns the Popen return value:

```
# Exec string
command = f"mpirun -np {cores} python {script_file}"

# Launch simulation
proc = subprocess.Popen(command, shell=True)

return proc
```

which can be used to coordinate control flow somewhere else.

simbilod, 2022-01-17:
See PR#149

Instead of using a special simulation file with hardcoded command line arguments as is done in the picwriter example, I write temporary Python scripts that are executed using mpi through Popen. This is much better to handle arbitrary **kwargs.

To do this I need to pickle the component object to disk so that I can load it in the generated python file. @joamatab can this cause any problem?

I didn't know what test to add for this, suggestions welcome.

## Example launching a single MPI simulation from Python

The function takes in a simulation "instance", which is a Dict whose keys are arguments of `write_sparameters_meep` and the values the associated values.

```
    """
    Parallel example
    """

    c1 = gf.c.straight(length=5)
    p = 3
    c1 = gf.add_padding_container(c1, default=0, top=p, bottom=p)

    instance_dict = {
        "component": c1,
        "run": True,
        "overwrite": True,
        "lazy_parallelism": True,
        "filepath": "instance_dict.csv",
    }

    proc = write_sparameters_meep_parallel(
        instance=instance_dict,
        cores=3,
        verbosity=True,
    )
```

`write_sparameters_meep_parallel` writes a temporary python file:

```
import pickle
from gdsfactory.simulation.gmeep import write_sparameters_meep

if __name__ == "__main__":

	with open("/home/simbil/Github/gdsfactory/gdslib/sp/temp/write_sparameters_meep_parallel.pkl", 'rb') as inp:
		component = pickle.load(inp)

	write_sparameters_meep(component = component,
		run = True,
		overwrite = True,
		lazy_parallelism = True,
		filepath = "instance_dict.csv",
	)
```
and executes it as 

```
    # Exec string
    command = f"mpirun -np {cores} python {script_file}"

    # Launch simulation
    proc = subprocess.Popen(command, shell=True)
```
The subprocess.Popen object is returned to allow for higher-level coordination of multiple simulations, as implemented below.

## Example launching a set of MPI simulations

I added another function that takes a list of instances. This is useful for parameter sweeps to avoid diminishing returns of large core numbers on single simulations. The user can always just provide a single instance to simulate in the list, and dedicate all cores to it.

```
    """
    Parallel example
    """

    c1 = gf.c.straight(length=5)
    p = 3
    c1 = gf.add_padding_container(c1, default=0, top=p, bottom=p)

    instance_dict = {
        "component": c1,
        "run": True,
        "overwrite": True,
        "lazy_parallelism": True,
        "filepath": "instance_dict.csv",
    }

    proc = write_sparameters_meep_parallel(
        instance=instance_dict,
        cores=3,
        verbosity=True,
    )

    """
    Parallel pools example
    """

    c1 = gf.c.straight(length=5)
    p = 3
    c1 = gf.add_padding_container(c1, default=0, top=p, bottom=p)

    c2 = gf.c.straight(length=4)
    p = 3
    c2 = gf.add_padding_container(c2, default=0, top=p, bottom=p)

    c1_dict = {
        "component": c1,
        "run": True,
        "overwrite": True,
        "lazy_parallelism": True,
        "filepath": "c1_dict.csv",
    }
    c2_dict = {
        "component": c2,
        "run": True,
        "overwrite": True,
        "lazy_parallelism": True,
        "filepath": "c2_dict.csv",
    }
    c3_dict = {
        "component": c2,
        "run": True,
        "overwrite": True,
        "lazy_parallelism": True,
        "resolution": 40,
        "port_source_offset": 0.3,
        "filepath": "c3_dict.csv",
    }

    # Instances
    instances = [
        c1_dict,
        c2_dict,
        c3_dict,
    ]

    write_sparameters_meep_parallel_pools(
        instances=instances,
        cores_per_instance=4,
        total_cores=10,
        delete_temp_files=False,
        verbosity=True,
    )
```

I want to use a total of 10 cores, and 4 cores per instance. So the code will run the first 2 together, and the third one afterwards.

The (non-MEEP) output is as follows (with verbosity True, there is some extra output):

```
Running parallel simulations over 3 instances
Using a total of 10 cores with 4 cores per instance
Tasks split amongst 2 pools with up to 2 instances each.
Task 0 of pool 0 is instance 0
Launching: mpirun -np 4 python /home/simbil/Github/gdsfactory/gdslib/sp/temp/write_sparameters_meep_parallel_0.py
Task 1 of pool 0 is instance 1
Launching: mpirun -np 4 python /home/simbil/Github/gdsfactory/gdslib/sp/temp/write_sparameters_meep_parallel_1.py
```
At this point, two 4-core MPI simulations are properly running:
[image]

Inside `write_sparameters_meep_parallel_pools`, we collected the proc objects and wait until they are all done:
```
for process in processes:
    process.wait()
```
When that is the case, the script moves on to pool 2:
```
Task 0 of pool 1 is instance 2
Launching: mpirun -np 4 python /home/simbil/Github/gdsfactory/gdslib/sp/temp/write_sparameters_meep_parallel_2.py
```
Which properly executes on 4 cores:
[image]

joamatab, 2022-01-17:
#149 closes this, thank you Simon!