gdsfactory/gdsfactory #176: Add support for Google's open source 130nm PDK manufacturable at SkyWater

mithro, 2022-02-02:
In 2020, Google in collaboration with SkyWater released a fully open source PDK for Skywater's 130nm process technology under an Apache 2.0 license. It can be found at http://github.com/google/skywater-pdk with documentation at https://skywater-pdk.readthedocs.io/ -- The announcement video can be found at https://www.youtube.com/watch?v=EczW2IWdnOM&t=152s

It would be awesome for gdsfactory to ship with support for this PDK! There are a number of efforts to make sure that KLayout is well supported in the PDK - see https://github.com/laurentc2/SKY130_for_KLayout and https://github.com/mabrains/sky130_klayout_pdk

FYI - SkyWater also does a lot of work in the photonics space.

mithro, 2022-02-02:
FYI - @proppy @QuantamHD

mithro, 2022-02-02:
BTW My work address is tansell@google.com

joamatab, 2022-02-02:
Hi Tim,
Thank you for your message

yes, I am a fan of your work and would love to extend gdsfactory to support analog and digital electronic flows with the skywater pdk. As far as i know gdsfactory and its siblings (phidl, gdspy and klayout) are used in photonics design and superconducting RF circuits, and could easily be extended to more fields.

I'm happy that foundries are starting to embrace an open source PDK model, so IC designers don't need to manually export their cells from complex proprietary GUI based flows. I love that gdsfactory pure code driven flow can diff the PDKs (including boolean diffs at GDS level), version control them and write tests. Also, If the PDK was described in python in the first place it would make things much easier to maintain :-)

how do you think it is better to support SKY130?
We've started writing plugins in gdsfactory, and Im still debating whether plugins should be in separate repos or inside the gdsfactory repo. What do you think?

Im looking forward to working with you on this project
Joaquin

proppy, 2022-05-23:
Isn't this already ongoing at https://github.com/gdsfactory/skywater130 ?

joamatab, 2022-08-10:
yes, 

things to do:

- [ ] add pyspice examples
- [ ] better electrical router. With some pathfinder algorithms
- [ ] add useful analog examples

joamatab, 2023-08-03:
We'll follow this issue here
 https://github.com/gdsfactory/skywater130

closed 2023-08-03 by joamatab