gdsfactory/gdsfactory #188: enable tighter metal routes

joamatab, 2022-02-17:
The current routing algorithm does not allow for small "kinks" in the wire.
More precisely, for the current wiring algorithms (gf.routing.get_route.getroute() and gf.routing.get_bundle.get_bundle()), if the wire needs to make an S-bend, the lateral displacement of the wire must be GREATER than the wire width. This makes tight wiring and/or very painful.

```
import gdsfactory as gf

if __name__ == "__main__":
    c = gf.Component("mzi_with_pads")
    c1 = c << gf.components.pad()
    c2 = c << gf.components.pad()

    c2.movex(30)
    c2.movey(200)

    port1 = c1.ports["e2"]
    port2 = c2.ports["e4"]

    route = gf.routing.get_route(
        port1,
        port2,
        cross_section=gf.cross_section.metal1,
        width=50,
        bend=gf.components.wire_corner,
    )
    c.add(route.references)
    c.show()
```

joamatab, 2022-02-17:
@flaport maybe your non-manhattan router could help with this?

photonicsflow, 2022-02-18:
It looks like the ports of your corner are on the edges. Moving them in the center of the metal corner would fix this type of issue

flaport, 2022-02-18:
@joamatab , I did not really build an auto router. I just put some effort into enabling routing with non-90° euler bends. Not sure how applicable this is here...

joamatab, 2022-02-19:
Both centering the port pads or increasing y solve this issue

```
import gdsfactory as gf

if __name__ == "__main__":
    c = gf.Component("mzi_with_pads")
    c1 = c << gf.components.pad()
    c2 = c << gf.components.pad()

    c2.movex(30)
    c2.movey(300)

    port1 = c1.ports["pad"]
    port2 = c2.ports["pad"]

    route = gf.routing.get_route(
        port1,
        port2,
        cross_section=gf.cross_section.metal1,
        width=50,
        bend=gf.components.wire_corner,
    )
    c.add(route.references)
    c.show()
```

```
import gdsfactory as gf

if __name__ == "__main__":
    c = gf.Component("mzi_with_pads")
    c1 = c << gf.components.pad()
    c2 = c << gf.components.pad()

    c2.movex(30)
    c2.movey(300)

    port1 = c1.ports["e2"]
    port2 = c2.ports["e4"]

    route = gf.routing.get_route(
        port1,
        port2,
        cross_section=gf.cross_section.metal1,
        width=50,
        bend=gf.components.wire_corner,
    )
    c.add(route.references)
    c.show()
```

closed 2022-02-19 by joamatab