gdsfactory/gdsfactory #191: tidy3d plugin not finding the right waveguide modes

joamatab, 2022-02-19:
The waveguide modes have wrong index and mode profiles

[image]

tylerflex, 2022-02-22:
Can you paste the output of `sim.json()`?
Also if you can get the mode solver, the output of the `mode_solver.json()`.
That would help debug this, because I can't reproduce.

joamatab, 2022-02-23:
sim.json
```
{"center": [0.0, 0.0, 0.0], "type": "Simulation", "size": [5.0, 4.5, 4.22], "grid_size": [0.02, 0.02, 0.02], "medium": {"name": null, "frequency_range": null, "type": "Medium", "permittivity": 1.0, "conductivity": 0.0}, "run_time": 1.0340486931104608e-12, "structures": [{"geometry": {"center": [0.0, 0.0, 0.0], "type": "Box", "size": [1.0000000200408773e+20, 1.0000000200408773e+20, 1.0000000200408773e+20]}, "medium": {"name": null, "frequency_range": null, "type": "Medium", "permittivity": 2.0736, "conductivity": 0.0}, "name": null, "type": "Structure"}, {"geometry": {"center": [-3.5, 2.4054832200211723e-16, 0.11], "type": "PolySlab", "axis": 2, "length": 0.22, "slab_bounds": [0.0, 0.22], "vertices": [[-1.5, 0.25], [-5.5, 0.2500000000000005], [-5.5, -0.2499999999999995], [-1.5, -0.25]]}, "medium": {"name": null, "frequency_range": null, "type": "Medium", "permittivity": 12.0409, "conductivity": 0.0}, "name": null, "type": "Structure"}, {"geometry": {"center": [3.5, -0.0, 0.11], "type": "PolySlab", "axis": 2, "length": 0.22, "slab_bounds": [0.0, 0.22], "vertices": [[1.5, -0.25], [5.5, -0.25], [5.5, 0.25], [1.5, 0.25]]}, "medium": {"name": null, "frequency_range": null, "type": "Medium", "permittivity": 12.0409, "conductivity": 0.0}, "name": null, "type": "Structure"}, {"geometry": {"center": [-0.0, -0.0, 0.11], "type": "PolySlab", "axis": 2, "length": 0.22, "slab_bounds": [0.0, 0.22], "vertices": [[-1.5, -0.25], [1.5, -0.25], [1.5, 0.25], [-1.5, 0.25]]}, "medium": {"name": null, "frequency_range": null, "type": "Medium", "permittivity": 12.0409, "conductivity": 0.0}, "name": null, "type": "Structure"}], "sources": [{"center": [-1.5, 0.0, 0.0], "type": "ModeSource", "size": [0.0, 2.5, 2.2199999999999998], "source_time": {"amplitude": 1.0, "phase": 0.0, "type": "GaussianPulse", "freq0": 193414489407062.47, "fwidth": 19341448940706.246, "offset": 5.0}, "name": null, "direction": "+", "mode_spec": {"num_modes": 1, "target_neff": null, "symmetries": [0, 0], "num_pml": [0, 0], "bend_radius": null, "bend_axis": null, "type": "ModeSpec"}, "mode_index": 0}], "monitors": [{"center": [0.0, 0.0, 0.11], "type": "FieldMonitor", "size": [5.0, 4.5, 0.0], "name": "field", "freqs": [193414489407062.47], "fields": ["Ex", "Ey", "Ez", "Hx", "Hy", "Hz"]}, {"center": [-1.3, -2.4492935982947065e-17, 0.0], "type": "ModeMonitor", "size": [0.0, 2.5, 2.2199999999999998], "name": "o1", "freqs": [193414489407062.47], "mode_spec": {"num_modes": 1, "target_neff": null, "symmetries": [0, 0], "num_pml": [0, 0], "bend_radius": null, "bend_axis": null, "type": "ModeSpec"}}, {"center": [1.5, 0.0, 0.0], "type": "ModeMonitor", "size": [0.0, 2.5, 2.2199999999999998], "name": "o2", "freqs": [193414489407062.47], "mode_spec": {"num_modes": 1, "target_neff": null, "symmetries": [0, 0], "num_pml": [0, 0], "bend_radius": null, "bend_axis": null, "type": "ModeSpec"}}], "pml_layers": [{"num_layers": 12, "parameters": {"sigma_order": 3, "sigma_min": 0.0, "sigma_max": 1.5, "type": "PMLParams", "kappa_order": 3, "kappa_min": 1.0, "kappa_max": 3.0, "alpha_order": 1, "alpha_min": 0.0, "alpha_max": 0.0}, "type": "PML"}, {"num_layers": 12, "parameters": {"sigma_order": 3, "sigma_min": 0.0, "sigma_max": 1.5, "type": "PMLParams", "kappa_order": 3, "kappa_min": 1.0, "kappa_max": 3.0, "alpha_order": 1, "alpha_min": 0.0, "alpha_max": 0.0}, "type": "PML"}, {"num_layers": 12, "parameters": {"sigma_order": 3, "sigma_min": 0.0, "sigma_max": 1.5, "type": "PMLParams", "kappa_order": 3, "kappa_min": 1.0, "kappa_max": 3.0, "alpha_order": 1, "alpha_min": 0.0, "alpha_max": 0.0}, "type": "PML"}], "symmetry": [0, 0, 0], "shutoff": 1e-05, "subpixel": true, "courant": 0.9, "version": "1.0.1"}
```

mode_solver has no `json()` method

tylerflex, 2022-02-23:
True. Let's see, could you send the `mode_solver.plane.json()`, `mode_solver.sim.json()` and `mode_solver.freq`?

tylerflex, 2022-02-23:
Actually, I was able to reproduce now, let me look into it.

tylerflex, 2022-02-23:
Momchil is looking into it, we think it's because you have two poly slabs that meet each other exactly at x=-1.5.  For a quick fix, either merge your poly slabs of the same material or move the source position just a bit +/- in x and let me know if that works.  The server side mode solver in production should not have issues with this so you probably notice normal looking fields in your simulation

momchil-flex, 2022-02-23:
Yep, it's exactly what Tyler said, and yeah I would expect that the simulation still works as expected. This is just an issue of the python epsilon getter and numerical precision when your source is sitting exactly at the interface between two structures.

joamatab, 2022-02-23:
sounds good Momchil and Tyler,

yes, the results look good, it's just the modes that do not look right,

in the meantime i can add offset the source to make sure that the source is not at the interface of both polySlabs

joamatab, 2022-02-23:
I added a `port_source_offset` parameter as a workaround to the issue

the modes look good now

[image]

thank you for your help and prompt response

closed 2022-02-23 by joamatab