gdsfactory/gdsfactory #220: gdsfactory seems to depends on multiple version of the `panel` package

proppy, 2022-03-08:
Installing with `python -m pip install gdsfactory` trigger the following install:
```
Collecting panel>=0.8.0
  Downloading panel-0.12.5-py2.py3-none-any.whl (12.9 MB)
     |████████████████████████████████| 12.9 MB 28.5 MB/s 
  Downloading panel-0.12.4-py2.py3-none-any.whl (12.9 MB)
     |████████████████████████████████| 12.9 MB 31.4 MB/s 
  Downloading panel-0.12.3-py2.py3-none-any.whl (12.8 MB)
     |████████████████████████████████| 12.8 MB 828 kB/s 
  Downloading panel-0.12.2-py2.py3-none-any.whl (12.8 MB)
     |████████████████████████████████| 12.8 MB 40.2 MB/s 
  Downloading panel-0.12.1-py2.py3-none-any.whl (12.8 MB)
     |████████████████████████████████| 12.8 MB 37.9 MB/s 
  Downloading panel-0.12.0-py2.py3-none-any.whl (9.9 MB)
     |████████████████████████████████| 9.9 MB 14.2 MB/s 
  Downloading panel-0.11.3-py2.py3-none-any.whl (9.0 MB)
     |████████████████████████████████| 9.0 MB 28.5 MB/s 
  Downloading panel-0.11.2-py2.py3-none-any.whl (9.0 MB)
     |████████████████████████████████| 9.0 MB 47.3 MB/s 
  Downloading panel-0.11.1-py2.py3-none-any.whl (8.8 MB)
     |████████████████████████████████| 8.8 MB 27.2 MB/s 
  Downloading panel-0.11.0-py2.py3-none-any.whl (8.8 MB)
     |████████████████████████████████| 8.8 MB 26.7 MB/s 
  Downloading panel-0.10.3-py2.py3-none-any.whl (8.1 MB)
     |████████████████████████████████| 8.1 MB 40.4 MB/s 
```
is that expected?

joamatab, 2022-03-09:
thank you @proppy 

I updated gdsfactory to work on colab

for some reason colab was not finding the klayout layermap which defines the layercolors 

so i made a default colormap in case it fails in the import

your colab should work now

```
!pip install gdsfactory --upgrade
!pip install holoviews bokeh
```

this was related to issue #221 

let me know if you find further possible improvements

joamatab, 2022-03-11:
closed by https://github.com/gdsfactory/gdsfactory/pull/222

closed 2022-03-11 by joamatab