gdsfactory/gdsfactory #221: Don't pin versions in setup.py

flaport, 2022-03-09:
Hey Joaquin,

Installing gdsfactory in an existing python environment is a pain. This is because all dependencies of gdsfactory are pinned to an exact version. In my opinion this is not a good system for a *library*, as we need to be able to place it in many different contexts.

In my opinion you should pin dependency versions in your requirements.txt but **NOT** in your setup.py.

i.e. requirements.txt is your 'recommended environment' whereas in setup.py you should have minimal dependencies such that your library can easily be installed in other environments. And yes, that might break things *sometimes* but it's the end user's choice to use/not use your requirements.txt.

Let me know what you think.

joamatab, 2022-03-09:
thank you Floris, let me know if it's fixed on your end or any possible improvements

flaport, 2022-03-10:
This is great! Thanks Joaquin :)

closed 2022-03-10 by joamatab