gdsfactory/gdsfactory #223: binder runs quite slow

joamatab, 2022-03-09:
I noticed that binder runs now quite slow, any ideas why?

@tvt173 
@proppy
@flaport

tvt173, 2022-03-10:
i had noticed the last time i ran it, it seems to be the bokeh plots taking a long time. not sure if matplotlib is any faster

joamatab, 2022-03-10:
also some bokeh notebooks went above Max Github 100MB limit so the documentation is failing to build

```
remote: warning: File notebooks/00_geometry.html is 63.09 MB; this is larger than GitHub's recommended maximum file size of 50.00 MB        
  remote: warning: File notebooks/01_references.html is 84.67 MB; this is larger than GitHub's recommended maximum file size of 50.00 MB        
  remote: warning: File notebooks/02_movement.html is 59.59 MB; this is larger than GitHub's recommended maximum file size of 50.00 MB        
  remote: warning: File notebooks/03_layers.html is 52.80 MB; this is larger than GitHub's recommended maximum file size of 50.00 MB        
  remote: warning: File notebooks/04_components_hierarchy.html is 57.17 MB; this is larger than GitHub's recommended maximum file size of 50.00 MB        
  remote: error: Trace: 61b45c20e015111ca395450b45e07f0ac5393050d67676609330d603ac89a94c        
  remote: error: See http://git.io/iEPt8g for more information.        
  remote: error: File notebooks/03_waveguides_paths_crossections.ipynb is 208.56 MB; this exceeds GitHub's file size limit of 100.00 MB        
  remote: error: File notebooks/01_references.ipynb is 171.81 MB; this exceeds GitHub's file size limit of 100.00 MB        
  remote: error: File notebooks/04_components_hierarchy.ipynb is 115.24 MB; this exceeds GitHub's file size limit of 100.00 MB        
  remote: error: File notebooks/03_waveguides_paths_crossections.html is 103.19 MB; this exceeds GitHub's file size limit of 100.00 MB        
  remote: error: File notebooks/00_geometry.ipynb is 128.37 MB; this exceeds GitHub's file size limit of 100.00 MB        
  remote: error: File notebooks/02_movement.ipynb is 121.25 MB; this exceeds GitHub's file size limit of 100.00 MB        
  remote: error: File notebooks/03_layers.ipynb is 103.72 MB; this exceeds GitHub's file size limit of 100.00 MB        
  remote: error: GH001: Large files detected. You may want to try Git Large File Storage - https://git-lfs.github.com./

```

how about we disable holoviews as the default plotter?
we can leave the `# gf.CONF.plotter = 'holoviews' ` so it's easy to turn on

tvt173, 2022-03-10:
yeah, seems reasonable

closed 2022-03-13 by joamatab