gdsfactory/gdsfactory #229: recognize hierarchical components in SiEPIC tools verification flow?

joamatab, 2022-03-12:
SiePIC tools working now. which is nice for verification (find unconnected pins, extract interconnect netlist ..)

See `gdsfactory/samples/pdk/fab_a.py`

How can we recognize hierarchical components?

[image]

thomasdorch, 2022-03-12:
Do the lower-level components need to be flattened? Removing the "absorb" calls in coupler_ring and coupler90 fixes this

[image]

joamatab, 2022-03-12:
I see, thank you Thomas

joamatab, 2022-03-16:
yes, that worked great, thank you Thomas,

closed 2022-03-16 by joamatab