gdsfactory/gdsfactory #245: document interconnect Lumerical Ansys gdsfactory plugin

joamatab, 2022-03-17:
it would be great having a notebook tutorial in the simulations section that shows how to use the interconnect plugin 

https://github.com/gdsfactory/gdsfactory/pull/237

thomasdorch, 2022-03-17:
In the works!

joamatab, 2022-03-17:
Awesome 😎

joamatab, 2022-06-21:
Hi Thomas,

thanks for your PR #469 

When i run the notebook on my end

[image]

[image]

@thomasdorch

thomasdorch, 2022-06-21:
The `failed to create element` sounds like the CML didn't install properly, is the EBeam library is shown under Design Kits?

thomasdorch, 2022-06-21:
Okay so I think this was happening because the EBeam CML wasn't getting installed due to `initialize_design_kit` finding the `interconnect` path in gdslib and not installing it. This should be fixed with [this PR](https://github.com/gdsfactory/gdsfactory/pull/472)

joamatab, 2022-08-13:
Awesome, thank you Thomas

closed 2022-08-13 by joamatab