gdsfactory/gdsfactory #247: foundry PDKs

joamatab, 2022-03-21:
one of the biggest limiting factors for a larger adoption of gdfactory is that we don't provide official PDK foundry support

what is the best way to share PDKs for different foundries?

Most fabs support cadence and synopsis PDKs at the moment, and it would be great to enable an open source alternative (which in my opinion is much nicer to use)
maybe you can also help us gain some traction by asking your partner fabs to work with us on this. Feel free to share and comment this post.

I see 3 options:

1. In the ideal case the foundry agrees to open source their PDK. See for example [google/skywater130 pdk](https://github.com/google/skywater-pdk) This will enable them to have a large, community built and maintained PDK. No NDAs required for accessing PDK technology needed for building chips.
2. A harder to maintain option, is to create a private Github repository, that the foundry controls access, and only users with valid NDA get granted permission to access
3. A final option, is that the foundry builds and maintains a gdsfactory PDK for their technology and they share it with their users. This is the least collaborative approach and does not allow users to share improvements. This is option is based on the current model and requires extra effort from the foundry to build and maintain the PDK. Therefore PDKs are hard to maintain, hard to track changes and hard to improve.

what do you think?

@quantamHD
@mithro
@thomasdorch

mithro, 2022-03-22:
FYI - @proppy @msaligane

Let's start with the option (1) for SKY130, the second fully open source process we plan to release this year and ASAP7 (non-manufacturable)? We don't even need any of the complexity around black boxed cells here. After that all already works, then we can then look at the more complicated option (2) and/or option (3).

alexsludds, 2022-05-01:
I can't disclose the PDK, but if it of interest to folks and I can put the AIM photonics layer stack in gdsfactory format for folks to use on their tapeouts.

joamatab, 2022-05-01:
Hi Alex,

thank you for the offer, currently most PDKs are distributed by foundries under NDA. 

The only one that does not require NDA at the moment is the skywater PDK, which was open sourced by Google and skywater on 2020, and the ubcpdk that Lukas uses on his EDX course.

The decision about open sourcing should come from the foundries, so if you don't want to be the only one maintaining that PDK please ask your foundry for advice

closed 2022-08-10 by joamatab