gdsfactory/gdsfactory #249: Connecting two labels in GDS layout

ratul619, 2022-03-29:
Hi Team , 

I have below requirement where i need to connect two "labels" in GDS : 

1. I have two design gds - lets say A.gds and B.gds - and they have pin A1,A2 and B1,B2 respectively. 
2. I need to create a top.gds comprising of A.gds and B.gds where i need to connect A1-B1 , A2-B2 with some layer without overlap. 

Can you please guide if its possible and how to do it ? 

Thanks & Regards

tvt173, 2022-03-30:
Trying to understand what you are saying and restate in a way that would be do-able in the code... 

You want to read in two gds files, and place them as two separate cells in a new top-level gds file.
Then you want to convert labels on each of the cells to ports, and route between them on some layer?

Is that correct? If not, you may want to draw a diagram to explain what you mean...

ratul619, 2022-03-30:
[image]

ratul619, 2022-03-30:
Lets take an example. There are two shapes on which there are two different TEXT labels created with layers met1.label and met2.label respectively and named as A1 and B1. How do i connect them with chosen layer number and shape (width) ? 

Also if there is any shape in between them - can it do a detour and connect ? Like my next diagram

ratul619, 2022-03-30:
[image]

ratul619, 2022-04-01:
Hi , Was the requirement clear or more information is needed ? 

Thanks

tvt173, 2022-04-01:
hi @ratul619 , sorry I've been busy with other things. this is definitely possible, let me point you to the relevant portions of the docs

first, you will want to read in those gds files and import them as components. there are functions to automatically convert those labels to ports on import, which you will need for routing later:
https://gdsfactory.github.io/gdsfactory/notebooks/05_GDS.html#GDS-input-+-port-pins

(a more basic primer on ports)
https://gdsfactory.github.io/gdsfactory/notebooks/00_geometry.html#Connect-ports

then you will want to route between them. `get_route_from_waypoints()` can be used to route around obstacles like this
https://gdsfactory.github.io/gdsfactory/routing.html#gdsfactory.routing.get_route.get_route_from_waypoints

hope this helps

ratul619, 2022-04-01:
Thanks @tvt173  - i will check these out !!

closed 2022-04-22 by joamatab