gdsfactory/gdsfactory #259: enable custom fonts in polygon text

tvt173, 2022-04-04:
it would be nice if the user could easily load a custom font to use for polygon text. 

currently the [text()](https://github.com/gdsfactory/gdsfactory/blob/master/gdsfactory/components/text.py) function references private variables for character glyphs, widths, and indents. it would be nice if it was easier to do this, maybe from a library of gds characters.

relevant to
heitzmann/gdspy/issues/207

joamatab, 2022-04-14:
@spauka worked on this for phidl

spauka, 2022-04-15:
Hi @tvt173,

The implementation for PHIDL is here https://github.com/amccaugh/phidl/blob/master/phidl/font.py

Font reading is performed using freetype-py, and once this is done you would need a curve implementation that supports cubic or quadratic bezier curves.

It would be reasonable straightforward to port to use GDSPY polygons/curves (which do support the right curves) rather than PHIDLs `Device`, although it's not something I would likely have the time to do, but let me know if I could help at all.

tvt173, 2022-04-15:
very cool, @spauka ! thanks for sharing. will re-share this on the original gdspy ticket

joamatab, 2022-10-07:
#743 

See docs here
https://gdsfactory.github.io/gdsfactory/components.html#text-freetype

closed 2022-10-07 by joamatab