gdsfactory/gdsfactory #266: confusing name for cross_section kwarg

joamatab, 2022-04-12:
many functions (straight, bend ...) have a cross_section kwarg

However most routing functions work only with CrossSection factories

we have 3 options:

1. Accept only cross_section_factory. In that case maybe we should rename `cross_section` to `cross_section_factory`
2. Accept only CrossSection and remove **kwargs
3. Accept both, and check if cross_section is callable or not

What do you think?
@tvt173 
@flaport

flaport, 2022-04-12:
I think 2. accepting only a CrossSection is too limiting. Sometimes a cross-section might change during routing.

On the flipside, 1 is maybe also not perfect. although you can probably solve pre-defined cross-sections with partials in that case...

Overall I think option 3 is best. However, what about accepting _both_ `cross_section` _and_ `cross_section_factory` and make both options mutually exclusive? I think that's the most explicit option.

joamatab, 2022-04-12:
sounds good, 

we can keep using option3

I started a branch to try out Option2. See branch https://github.com/gdsfactory/gdsfactory/tree/472_cross_section_only

but it could make people confused

tvt173, 2022-04-13:
the major downsides of the proposed method are that

1. You must create/maintain/activate the registry of cross sections. We should put careful thought into how this is done.
2. You must remember to call the `get_cross_section()` method within your component before using it. But I think this is pretty simple to do in practice, and if your args are type annotated, modern IDE's like Pycharm will yell at you if you don't do this before trying to use the passed cross_section parameter as a CrossSection.

Overall, I think benefits outweigh the downsides, but let me know what you think! Another cool side effect is you can supply default values which are _strings_... the user can then create their own registry of cross sections, which will override what gets dynamically fetched when using their own PDK if they comply with the same conventions for naming 'strip' and 'rib' cross sections, etc.!

joamatab, 2022-04-13:
Hi Troy,

I like your idea of the cross_section spec, as a way to remove kwargs from many components and functions

joamatab, 2022-04-14:
https://github.com/gdsfactory/gdsfactory/pull/272

implements get_cross_section and get_component

Im going to see if we can remove **kwargs in other PR

and maybe make sure cross_section has more explicit routing params

closed 2022-04-21 by joamatab