gdsfactory/gdsfactory #282: Serialization error: cannot represent an object with type Section

tvt173, 2022-04-22:
currently if you have a component with a parameter of type CrossSection, it cannot be serialized.

simple example:

try serializing the settings for the following component, i.e. by generating a regression test for it
```python
@gf.cell
def demonstrate_bad_cross_section_setting(cross_section=strip()) -> gf.Component:
    c = gf.components.straight(cross_section=cross_section)
    return c
```

i think this used to work, but maybe this code was removed when handling of cross_sections was refactored?

joamatab, 2022-04-22:
Hi Troy,

i added this test and it worked great on my side

```

import gdsfactory as gf
from gdsfactory.cross_section import strip

@gf.cell
def demo_cross_section_setting(cross_section=strip()) -> gf.Component:
    c = gf.components.straight(cross_section=cross_section)
    return c

def test_settings(data_regression, check: bool = True) -> None:
    """Avoid regressions when exporting settings."""
    component = demo_cross_section_setting()
    data_regression.check(component.to_dict())

if __name__ == "__main__":
    # c = demo_cross_section_setting()
    # d = c.to_dict()
    # c.show()
    test_settings()
```

joamatab, 2022-04-22:
see https://github.com/gdsfactory/gdsfactory/blob/master/gdsfactory/tests/test_serialize.py

closed 2022-04-26 by joamatab