gdsfactory/gdsfactory #301: Import from .gds + .yml leads to layers being lists instead of tuples

Jan-David-Black, 2022-04-27:
When running `c = gf.import_gds()`on a component previously exported using `c.write_gds_with_metadata()` and inspecting `c.ports[...].layer` I find the layer to be a list (e.g. [1,0] instead of (1,0)). This in itself does not pose a big problem. But when successively using `gm.write_sparameters_meep` on the given component for example, it complains about not being a tuple but a list instead.

This could be fixed by changing this [line](https://github.com/gdsfactory/gdsfactory/blob/c0e7bcff7c2ccfcf28953f2d5b8ef40c73de9783/gdsfactory/read/import_gds.py#L197) to convert from list to tuple

joamatab, 2022-04-27:
awesome @Jan-David-Black 

see https://github.com/gdsfactory/gdsfactory/pull/305

let me know what you think
Joaquin

closed 2022-04-27 by joamatab

Jan-David-Black, 2022-04-28:
Great :)