gdsfactory/gdsfactory #315: Settings for imported gds cells without metadata file

thomasdorch, 2022-04-28:
Currently, import_gds only adds Settings to imported cells if there is a metadata file available. Because of this, the cells generated this way cause an exception when get_netlist is called. It would be nice to have a way to add Settings to these cells without having to call the cell decorator.

joamatab, 2022-04-28:
Hi Thomas,

can you use info?

[image]

thomasdorch, 2022-04-28:
Yes, info works for what I'm doing. The issue is that get_netlist doesn't copy over the info and throws an error if the component's settings dict is empty. 

PR #316 fixes this

joamatab, 2022-04-28:
Awesome! thank you Thomas

i merged and released 5.3.4

https://github.com/gdsfactory/gdsfactory/pull/320

closed 2022-04-28 by thomasdorch