gdsfactory/gdsfactory #322: Issue with add_fiber_array

alexsludds, 2022-04-29:
Hi folks,
This is probably just an issue with my understanding of the workflow for gdsfactory, but I am using the gf.routing.add_fiber_array to create some test structures and they are not being added.

I have added example code here of what I am doing.

```
import gdsfactory as gf
gf.config.set_plot_options(show_subports=False)
import numpy as np
from scipy.interpolate import interp1d

#First we create an imbalanced interferometer
@gf.cell 
def hochberg_1x2MMI():
    #Same MMI as in https://opg.optica.org/oe/fulltext.cfm?uri=oe-21-1-1310&id=248418
    def mmi_widths(t):
        widths = np.array([0.5,0.5,0.6,0.7,0.9,1.26,1.4,1.4,1.4,1.4,1.31,1.2,1.2])
        xold = np.linspace(0,1,num=len(widths))
        xnew = np.linspace(0,1,num=100)
        f = interp1d(xold,widths,kind='cubic')
        return f(xnew)

    c = gf.Component("hochberg_1x2MMI")

    P = gf.path.straight(length=2, npoints=100)
    X = gf.CrossSection(width=mmi_widths, offset=0, layer=(1, 0))
    c = gf.path.extrude(P, cross_section=X)
    c.unlock()

    #Add an input rectangular section and output section
    input_block = c << gf.components.rectangle((0.25,0.5),layer=(1,0)).move((-0.25,-0.25))

    #Two output ports
    top_output_block = c << gf.components.rectangle((0.25,0.5),layer=(1,0)).move((2,0.6-0.5))
    bottom_output_block = c << gf.components.rectangle((0.25,0.5),layer=(1,0)).move((2,-0.6 -0.0))

    c.add_port("input",port=input_block.ports['e1'])
    c.add_port("top_output",port=top_output_block.ports['e3'])
    c.add_port("bottom_output",port=bottom_output_block.ports['e3'])
    return c

@gf.cell
def imbalanced_interferometer_1x2mmi():
    interferometer_length = 60
    c = gf.Component("hochberg_1x2MMI_test_structure")
    mmi1 = c << hochberg_1x2MMI().move((0,0))
    mmi2 = c << hochberg_1x2MMI().mirror().move((interferometer_length,0))

    route = gf.routing.get_route(mmi1.ports["bottom_output"], mmi2.ports["bottom_output"])
    c.add(route.references)

    port1 = mmi1.ports["top_output"]
    port2 = mmi2.ports["top_output"]
    imbalance_length = 200
    route = gf.routing.get_route_from_steps(
        port1=port1,
        port2=port2,
        bend = gf.components.bend_circular,
        steps=[
            {"x": port1.x, "y": port1.y},
            {"x": port1.x + 10, "y": port1.y + 0},
            {"x": port1.x + 10, "y": port1.y + imbalance_length//2},
            {"x": port2.x -10, "y": port2.y + imbalance_length//2},
            {"x": port2.x -10, "y": port2.y + 0},
            {"x": port2.x, "y": port2.y}
        ],
    )
    c.add(route.references)
    c.add_port("o1", port=mmi1.ports["input"])
    c.add_port("o2", port=mmi2.ports["input"])
    return c

c = imbalanced_interferometer_1x2mmi()

packaged_component = gf.routing.add_fiber_array(component=c)

gf.show(packaged_component)
packaged_component.plot()

packaged_component.write_gds("1x2_mmi_test_structure")
```

Best,
-Alex

joamatab, 2022-04-29:
Hi Alex,

the rectangle adds ports with `port_type = 'electrical'` by default

```
import gdsfactory as gf

gf.config.set_plot_options(show_subports=False)
import numpy as np
from scipy.interpolate import interp1d

# First we create an imbalanced interferometer
@gf.cell
def hochberg_1x2MMI():
    # Same MMI as in https://opg.optica.org/oe/fulltext.cfm?uri=oe-21-1-1310&id=248418
    def mmi_widths(t):
        widths = np.array(
            [0.5, 0.5, 0.6, 0.7, 0.9, 1.26, 1.4, 1.4, 1.4, 1.4, 1.31, 1.2, 1.2]
        )
        xold = np.linspace(0, 1, num=len(widths))
        xnew = np.linspace(0, 1, num=100)
        f = interp1d(xold, widths, kind="cubic")
        return f(xnew)

    c = gf.Component("hochberg_1x2MMI")

    P = gf.path.straight(length=2, npoints=100)
    X = gf.CrossSection(width=mmi_widths, offset=0, layer=(1, 0))
    c = gf.path.extrude(P, cross_section=X)
    c.unlock()

    # Add an input rectangular section and output section
    input_block = c << gf.components.rectangle((0.25, 0.5), layer=(1, 0), port_type='optical').move(
        (-0.25, -0.25)
    )

    # Two output ports
    top_output_block = c << gf.components.rectangle((0.25, 0.5), layer=(1, 0), port_type='optical').move(
        (2, 0.6 - 0.5)
    )
    bottom_output_block = c << gf.components.rectangle((0.25, 0.5), layer=(1, 0), port_type='optical').move(
        (2, -0.6 - 0.0)
    )

    c.add_port("input", port=input_block.ports["o1"])
    c.add_port("top_output", port=top_output_block.ports["o3"])
    c.add_port("bottom_output", port=bottom_output_block.ports["o3"])
    return c

@gf.cell
def imbalanced_interferometer_1x2mmi():
    interferometer_length = 60
    c = gf.Component("hochberg_1x2MMI_test_structure")
    mmi1 = c << hochberg_1x2MMI().move((0, 0))
    mmi2 = c << hochberg_1x2MMI().mirror().move((interferometer_length, 0))

    route = gf.routing.get_route(
        mmi1.ports["bottom_output"], mmi2.ports["bottom_output"]
    )
    c.add(route.references)

    port1 = mmi1.ports["top_output"]
    port2 = mmi2.ports["top_output"]
    imbalance_length = 200
    route = gf.routing.get_route_from_steps(
        port1=port1,
        port2=port2,
        bend=gf.components.bend_circular,
        steps=[
            {"x": port1.x, "y": port1.y},
            {"x": port1.x + 10, "y": port1.y + 0},
            {"x": port1.x + 10, "y": port1.y + imbalance_length // 2},
            {"x": port2.x - 10, "y": port2.y + imbalance_length // 2},
            {"x": port2.x - 10, "y": port2.y + 0},
            {"x": port2.x, "y": port2.y},
        ],
    )
    c.add(route.references)
    c.add_port("o1", port=mmi1.ports["input"])
    c.add_port("o2", port=mmi2.ports["input"])
    return c

if __name__ == "__main__":
    c = imbalanced_interferometer_1x2mmi()
    # print(c.ports)
    # c.show()

    packaged_component = gf.routing.add_fiber_single(component=c)
    packaged_component.show()

```

alexsludds, 2022-05-01:
Thanks for explaining Joaquin! That makes sense.
I'll remove the rectangles from my cell and replace them.
Best,
-Alex

closed 2022-05-01 by alexsludds