gdsfactory/gdsfactory #330: watch files

joamatab, 2022-05-02:
- `gf yaml watch` only watches on file

it would be great adding a file watcher that monitor a list of files and displays in klayout the latest changed file

watchdog can do this

tvt173, 2022-05-02:
example with watchdog:

```python
def watch_project_files(project_path):
    event_handler = PatternMatchingEventHandler(patterns=["*.yml", "*.yaml"],
                                                ignore_patterns=[],
                                                ignore_directories=True)
    event_handler.on_any_event = on_modified

    observer = Observer()
    observer.schedule(event_handler, str(project_path), recursive=True)
    observer.start()
    try:
        while True:
            time.sleep(50)
    except KeyboardInterrupt:
        observer.stop()
    observer.join()
```

joamatab, 2022-05-02:
Hi Troy,

thank you

See version released version 5.4.0 with the watchdog

https://github.com/gdsfactory/gdsfactory/pull/337

And how the registration happens on the latest ubcpdk

https://github.com/gdsfactory/ubc/pull/29

For hierarchical YAML files, how do you reload dependencies?

tvt173, 2022-05-02:
> gdsfactory/ubc#29 <https://github.com/gdsfactory/ubc/pull/29>

The simplest way that's still effective is to track which pic is currently
in focus (i.e. in the viewer). Let's call that the current pic.

Then, when you get an event that a pic.yml file changed, if it is *only*
the file with the current pic, only invalidate the cache for instantiations
of that pic, and reload all factories for all changed functions. Otherwise,
clear the *entire* cache and reload all factories.

That way, if you are only editing the current pic (most common case) you
will have very fast response time because only the top level will need to
be reloaded. All subcells will be pulled from cache. If you change a
subcell, you will need to wait a bit longer because everything will be
reloaded, but at least it will still work.

joamatab, 2022-05-02:
sounds like a good idea,

how should we pass the settings to the YAML defined components?

tvt173, 2022-05-02:
Referencing yaml components from other components should be just like any
other component... Use a component spec with the component name and
settings.

For using parameters within the yaml, you can define a p-cell with a Jinja
template, which is pretty powerful but has the drawback of sometimes being
hard to debug and leaving pure yaml syntax. But if you want, I can probably
share.

tvt173, 2022-05-03:
hi @joamatab I'm out on vacation now, so please give me a few days to look into this properly

joamatab, 2022-05-03:
sounds good,

as far as i know your team is the main user of the netlist driven flow so i want to make sure it works well for you