gdsfactory/gdsfactory #331: run simulations on a server

joamatab, 2022-05-02:
it would be great to integrate a tool with the simulation plugins that lets you send simulations over a server

something like
https://snakemake.readthedocs.io/en/stable/

simbilod, 2022-05-06:
Snakemake is great, since you can just setup rules (shell commands that have files as inputs, and that can output new files). 

We could make a generic template for simulations, that looks up in the config file for which simulator to use, which components to simulate, how many processes to use, etc.

Remote server execution is straighforward once we have this file. Since gdsfactory and simulation backend is on mamba (at least for the open-source stuff), we can have so when executing on a remote server, the proper environment is setup. If the server environment has access to a job scheduler like SLURM, we could make a flag for that.

joamatab, 2022-05-07:
sounds great Simon,

this would be super cool to present at the meep conference at the end of july at MIT  :-)

@proppy has done something similar with papermill
@flaport  too

simbilod, 2022-05-08:
Cool, I'll take a look!

I'm trying the simplest thing with, just rewriting a gdsfactorysimulation  script as a Snakemake rule:

```
import gdsfactory as gf
import gdsfactory.simulation.gmeep as gm

rule waveguide_MEEP:
    output: filepath = "/home/simbil/debugging/snakemake_test.csv"
    run:
      c = gf.components.straight(length=5)
      df = gm.write_sparameters_meep_mpi(c, 
                                      resolution=20, 
                                      filepath=output.filepath, 
                                      dirpath="/home/simbil/debugging/",
                                      cores=2,)
```
It runs identically to its script version when calling `snakemake --cores 2`, so success!

Now I'm trying to get the containerized version (could also be conda) to work for server/cloud deployment. I needed to do some hacking, because the Snakemake `run` command doesn't work with containers (something about global python variables), but anyway we have:

```
rule waveguide_MEEP:
    container: 
      "docker://joamatab/gdsfactory:latest"
    output: filepath = "/home/simbil/debugging/snakemake_test.csv"
    shell:
      r"""
      python -c "import gdsfactory as gf
import gdsfactory.simulation.gmeep as gm
c = gf.components.straight(length=5)
df = gm.write_sparameters_meep_mpi(c, 
                                resolution=20, 
                                filepath='{output.filepath}', 
                                dirpath='/home/simbil/debugging/',
                                cores=2,)
"
      """
```

However, it seems the container version does not have proper access right:

```
>> snakemake --cores 2 --use-singularity
Building DAG of jobs...
Using shell: /usr/bin/bash
Provided cores: 2
Rules claiming more threads will be scaled down.
Job stats:
job               count    min threads    max threads
--------------  -------  -------------  -------------
waveguide_MEEP        1              1              1
total                 1              1              1

Select jobs to execute...

[Sat May  7 22:51:09 2022]
rule waveguide_MEEP:
    output: /home/simbil/debugging/snakemake_test.csv
    jobid: 0
    resources: tmpdir=/tmp

Activating singularity image /home/simbil/Github/gdsfactory/docs/notebooks/plugins/snakemake/.snakemake/singularity/abf2374502906de677e519237da349d8.simg
2022-05-07 22:51:10.944 | INFO     | gdsfactory.config:<module>:52 - Load '/opt/conda/lib/python3.8/site-packages/gdsfactory' 5.5.9
Using MPI version 4.0, 1 processes
2022-05-07 22:51:14.263 | INFO     | gdsfactory.simulation.gmeep:<module>:28 - Meep '1.23.0' installed at ['/opt/conda/lib/python3.8/site-packages/meep']
Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/pathlib.py", line 1287, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/opt/conda/lib/python3.8/site-packages/gdslib/sp/temp'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/pathlib.py", line 1287, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/opt/conda/lib/python3.8/site-packages/gdslib/sp'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 4, in <module>
  File "pydantic/decorator.py", line 40, in pydantic.decorator.validate_arguments.validate.wrapper_function
    from contextlib import _GeneratorContextManager
  File "pydantic/decorator.py", line 134, in pydantic.decorator.ValidatedFunction.call
    
  File "pydantic/decorator.py", line 201, in pydantic.decorator.ValidatedFunction.execute
    Fix args and kwargs to be consistent with the signature
  File "/opt/conda/lib/python3.8/site-packages/gdsfactory/simulation/gmeep/write_sparameters_meep_mpi.py", line 144, in write_sparameters_meep_mpi
    temp_dir.mkdir(exist_ok=True, parents=True)
  File "/opt/conda/lib/python3.8/pathlib.py", line 1291, in mkdir
    self.parent.mkdir(parents=True, exist_ok=True)
  File "/opt/conda/lib/python3.8/pathlib.py", line 1291, in mkdir
    self.parent.mkdir(parents=True, exist_ok=True)
  File "/opt/conda/lib/python3.8/pathlib.py", line 1287, in mkdir
    self._accessor.mkdir(self, mode)
OSError: [Errno 30] Read-only file system: '/opt/conda/lib/python3.8/site-packages/gdslib'

Elapsed run time = 0.0300 s
[Sat May  7 22:51:14 2022]
Error in rule waveguide_MEEP:
    jobid: 0
    output: /home/simbil/debugging/snakemake_test.csv
    shell:
        
      python -c "import gdsfactory as gf
import gdsfactory.simulation.gmeep as gm
c = gf.components.straight(length=5)
df = gm.write_sparameters_meep_mpi(c, 
                                resolution=20, 
                                filepath='/home/simbil/debugging/snakemake_test.csv', 
                                dirpath='/home/simbil/debugging/',
                                cores=2,)
"
      
        (one of the commands exited with non-zero exit code; note that snakemake uses bash strict mode!)

Shutting down, this might take some time.
Exiting because a job execution failed. Look above for error message
Complete log: .snakemake/log/2022-05-07T225109.018390.snakemake.log
```

@joamatab **is there a way to build the container so that it has write access to itself (if that is the problem?)?**

joamatab, 2022-05-09:
I think the issue is that you need to clone the sparameters folder from a different repo `gdslib`

`make gdslib` from gdsfactory should download it

im not super happy with storing Sparameters in a separate repo, and may be nicer to set up differently

- using DVC https://dvc.org/ and a Google cloud storage bucket
- using an SQL database

but so far is the simplest solution

simbilod, 2022-05-09:
Actually I found out about bind points for snakemake+containers, will try that to make gdsfactory run in singularity to write to the user filesystem

https://stackoverflow.com/questions/52742698/snakemake-docker-example-how-to-use-volumes

joamatab, 2022-05-12:
GCP announced https://cloud.google.com/run/docs/overview/what-is-cloud-run#jobs

we could add a silicon photonics labrad demo
https://github.com/GoogleCloudPlatform/rad-lab/tree/main/modules/silicon_design

where we run sims using meep on the cloud

@smartalech
@oskooi

joamatab, 2022-05-12:
https://github.com/NanoComp/meep/issues/2070

simbilod, 2022-05-13:
Where in the cloud do you want to run sims? I have a minimal example of running MEEP via snakemake and the most up-to-date MEEP container.

Snakefile:
```
rule MEEP:
    threads: 1
    container:
      "docker://mochen4/mpdocker:latest"
    output: "out.txt"
    shell:
      "python wvg-src.py > {output[0]}" 
```
Shell command: `snakemake --cores 1 --use-singularity`

with `wvg-src.py ` containing some MEEP script (still haven't looked into bindings for `gdsfactory `container file accesses).

This runs well locally. Snakemake offers built-in ways to run this elsewhere, for instance if you have a kubernetes cluster setup somewhere you just add `--kubernetes $REMOTE` flag, and there are also default cloud options `--default-remote-provider {S3,GS,FTP,SFTP,S3Mocked,gfal,gridftp,iRODS,AzBlob,XRootD}`

joamatab, 2022-08-13:
I think running it on GCP it would be great, that way we can run on demand large experiments on the cloud both from your local computer or from a notebook environment such as [radlab](https://github.com/GoogleCloudPlatform/rad-lab/tree/main/modules/silicon_design)

@simbilod 
@proppy
@mithro
@HelgeGehring

joamatab, 2022-11-21:
https://github.com/flexcompute/tidy3d/tree/develop/tidy3d

@SkandanC 
@dusandinho

joamatab, 2023-09-13:
closing this 

feel free to reopen it in https://github.com/gdsfactory/gplugins

closed 2023-09-13 by joamatab