gdsfactory/gdsfactory #363: complex indices for materials in write_sparameters_lumerical

Jan-David-Black, 2022-05-06:
**Is your feature request related to a problem? Please describe.**
Cannot use (n,k) Materials in Lumerical in the current state

**Describe the solution you'd like**
Enable passing a complex index to `write_sparameters_lumerical` (thru `material_name_to_lumerical`). This could be done by adding the following snippet to handle the complex index: 
``` python
elif isinstance(material, complex):
            mat = s.addmaterial("(n,k) Material")
            s.setmaterial(mat, "Refractive Index", material.real)
            s.setmaterial(mat, "Imaginary Refractive Index", material.imag)
            s.setnamed(layername, "material", mat)
```

Note that setting the material in lumerical depending whether a str, int or float is used is repeated twice, once for the cladding and once for the layers. It might make sense to refactor that into a function.

joamatab, 2022-05-06:
Hi @Jan-David-Black ,

yes, that's a great idea, thank you for sharing,

How does this new branch work for you?
https://github.com/gdsfactory/gdsfactory/tree/559 

I made a new function set_material() as you suggested

Jan-David-Black, 2022-05-06:
Looks Great! I am sorry I only throw stuff in here and don't just open a pull request, with everything set up properly. At some point I will find the time configure a "build environment" so that I can contribute more efficiently!

Jan-David-Black, 2022-05-06:
What was the problem with using complex instead of a tuple of floats?

joamatab, 2022-05-06:
now it can do both tuples and complex refractive index

[image]

closed 2022-05-07 by joamatab

Jan-David-Black, 2022-05-09:
> now it can do both tuples and complex refractive index

ah understood :)