gdsfactory/gdsfactory #368: compute bend loss

joamatab, 2022-05-07:
How can we calculate mode overlap to calculate the bend loss?

https://gdsfactory.github.io/gdsfactory/notebooks/plugins/tidy3d/01_tidy3d_modes.html

@simbilod 
@flaport

simbilod, 2022-05-08:
If your cross section is xy, shouldn`t `integrand = (
    np.conj(mode1_Ey) * mode2_Hz
    - np.conj(mode1_Ez) * mode2_Hy
    + mode2_Ey * np.conj(mode1_Hz)
    - mode2_Ez * np.conj(mode1_Hy)
)`

use x and y fields?

simbilod, 2022-05-08:
In https://github.com/gdsfactory/gdsfactory/blob/master/gdsfactory/simulation/modes/overlap.py I use the x-projection of the cross product leading to y and z components because x is the propagation direction

joamatab, 2022-05-08:
How about?

```
import numpy as np

index = 0 # TE mode

mode1_Ey = strip.Ey[index]
mode1_Ex = strip.Ex[index]
mode1_Hx = strip.Hx[index]
mode1_Hy = strip.Hy[index]

mode2_Ey = strip_bend.Ey[index]
mode2_Ex = strip_bend.Ex[index]
mode2_Hx = strip_bend.Hx[index]
mode2_Hy = strip_bend.Hy[index]
                
integrand = (
    np.conj(mode1_Ex) * mode2_Hy
    - np.conj(mode1_Ey) * mode2_Hx
    + mode2_Ex * np.conj(mode1_Hy)
    - mode2_Ey * np.conj(mode1_Hx)
)

integral = np.trapz(np.trapz(integrand, axis=0), axis=0)
abs(integral)
```

simbilod, 2022-05-08:
I think so, does it work? I don`t have a tidy3D account, so I don't know if I can run that notebook

joamatab, 2022-05-09:
yes, it seems to be working

[image]

The mode solver, runs locally, so you shouldn't need an account to run

flaport, 2022-05-09:
Hey Awesome!

Wit a tiny bit more work you effectively have an EME solver now!

Jan-David-Black, 2022-05-11:
> There are several mechanisms for loss in bends. (1) Scattering losses and substrate leakage: for the short distances considered in bends, these losses are small. As we will find, these losses are small for bend radii <10 μm. (2) Radiative loss: in highly confined waveguides, especially for TE modes in strip waveguides, radiation is typically small. It does contribute in rib waveguides for bend radii <5 μm and for TM-polarized modes. (3) Mode mismatch loss: this is the largest source of loss in bends. It is due to the imperfect mode overlap between the straight and the bent waveguides. This leads to scattering at the abrupt radius transition regions (start and end of fixedradius bends). There are several methods of decreasing mode-mismatch loss: (a) to laterally offset the straight waveguide relative to the bent waveguide in order to obtain a better mode overlap [29]; (b) to vary the curvature continuously, rather than abruptly; for example a 90◦ bend with an effective radius of 20 μm, where curvature changes from zero to 1/15 μm−1 then back to zero [30]. This has the added benefit that the bend does not excite higher order modes in the waveguide. See Figures 10.6, 10.7, and 10.8, for more details.

From Chrostowski, Hochberg: Silicon Photonics Design
Thus it seams that the radiation losses end up being relatively irrelevant for most cases

joamatab, 2022-05-11:
yes, in the notebook we only computed the losses coming from the mode missmatch from the bend into a straight waveguide.

closed 2022-05-15 by joamatab