gdsfactory/gdsfactory #384: Inform About incompatibility with python<3.7

Jan-David-Black, 2022-05-12:
**Describe the bug**
When installing on python 3.6 (I assume this holds true for smaller versions also) using pip:
```
pip install gdsfactory
```
the installation fails silently. Afterwards pip lists gdsfactory as an installed module, which however cannot be imported in python. In `site-packages` only a `.dist-info`is added but the source code itself is missing.

**To Reproduce**
Install gdsfactory in an environment wit python 3.6

**Expected behavior**
Fail to install or at least warn about incompatibility

**Suggested fix**
Have not looked into possible fixes yet

joamatab, 2022-05-13:
python3.6 has reached end of life

We have some options here for dealing with python3.6 installs:

1. we keep it as it is (installs an old version) of gdsfactory
2. we delete if all versions from 1.1.1 to 2.4.9 from pypi
3. we bring back 3.6 compatibility https://github.com/asottile-archive/future-annotations

What do you think?

joamatab, 2022-05-15:
I like options 1 and 2

@tvt173 
@flaport

flaport, 2022-05-15:
I think it's never a good idea to remove legacy versions to be honest. So I would be against option 2. Just go for 1 and list clearly in the Readme that gdsfactory is python 3.7+.

joamatab, 2022-05-15:
yes, 

we already say that on the docs installation section
https://gdsfactory.github.io/gdsfactory/notebooks/_2_klayout.html

Jan-David-Black, 2022-05-15:
Absolutely, I think, that the statement is already present in the docs. I was unsure about the degree of incompatibility to 3.6 however. Maybe we could add a sentence stating, that 2.4.9 was the last version supported by python 3.6. And that pip will select this version for py 3.6 environments, which is fundamentally and structurally different to the latest version.

tvt173, 2022-05-16:
sounds good to me, and seems to be the [standard official recommendation](https://packaging.python.org/en/latest/guides/dropping-older-python-versions/#dropping-a-python-release)

closed 2022-05-17 by joamatab