gdsfactory/gdsfactory #388: update to latest simphony

joamatab, 2022-05-15:
See https://github.com/gdsfactory/gdsfactory/tree/simphony_latest2

@SkandanC 

Still some test failing

SkandanC, 2022-05-15:
Hello @joamatab,

I took a look at the branch and ran pytest. I see which tests are failing now.

I also tried to install the .dat files from the `gdslib` repo through `install.bat` as you had suggested, but some files are still not available (such as the one in the attached screenshot).

[image]

My previous PR fixes most of the failing tests, except the ones with missing .dat files. There was also a test for the `coupler` component which was failing; the reason for this was that Simphony's components take units in m, unlike SiPANN which takes units in nm.

Please let me know if I have missed any failing tests that I haven't addressed.

joamatab, 2022-05-17:
Hi Skandan,

even in your PR the tests were not passing
https://github.com/gdsfactory/gdsfactory/runs/6428031102?check_suite_focus=true
so i had to revert your changes back after merging them,

make sure you checkout the branch `data` on gdslib repo

This is what i see when i run pytests on the `simphony_latest` branch

[image]

joamatab, 2022-05-17:
i made some progress but still not working

https://github.com/gdsfactory/gdsfactory/commit/5a794d35b926be5a888920ec9621c2d46649b98d

What are the advantages of the new simphony 0.6.1 versus the 0.3.0?

@austp
@sequoiap
@SkandanC

SkandanC, 2022-05-18:
Hello @joamatab,

The reason my last PR was failing was because the simphony version gdsfactory was installing is very old. Changing that to the newest version would fix that. 

[image]

I managed to find the .dat files and I am delighted to say that fixed more of the failing tests. Now I have 3-4 more tests failing on the `works-with-simphony-latest` branch, and am working on it now. 

Regarding the differences between the older and newer versions of Simphony, Austen and Seqouia are more qualified to answer that since they know more about the older versions.

SkandanC, 2022-05-18:
Hello @joamatab,

The reason my last PR was failing was because the simphony version gdsfactory was installing is very old. Changing that to the newest version would fix that. 

[image]

I managed to find the .dat files and I am delighted to say that fixed more of the failing tests. Now I have 3-4 more tests failing on the `works-with-simphony-latest` branch, and am working on it now.

joamatab, 2022-05-18:
Hi Skandan,

Sounds good, I met professor Camacho yesterday and he mentioned that there is a time domain solver coming soon in simphony, so it makes sense to upgrade to the latest simphony. Let me know when the tests are passing and I'm happy to merge your changes and upgrade to the latest version

SkandanC, 2022-05-29:
Hello @joamatab,

I have managed to fix all but 1 error and have opened a PR with my fixes. 

[image]

joamatab, 2022-05-31:
Hi Skandan,

i fixed that test but there are still other failing

[image]

can you fix this branch? 
https://github.com/gdsfactory/gdsfactory/tree/simphony_latest3

SkandanC, 2022-05-31:
Hello Joaquin, 

The errors arise from SiPANN where a specific set of lines cause this issue. 
[image]
As you can see in the image above, line 465 is the culprit. With this, I am able to reproduce your errors.

[image]

Changing the line to `if not all(...` fixes the issue.

I will let Austen and Seqouia know about this, and we will fix this as early as we can.

Let me know if you have any more comments.

SkandanC, 2022-06-01:
The issue has been fixed.

joamatab, 2022-06-10:
Can you make a new release for simphony with the new fixes?

joamatab, 2022-06-12:
Awesome, can you send a new PR after @seqouiap and @autsp release a new sipann?

SkandanC, 2022-06-12:
> Awesome, can you send a new PR after @seqouiap and @autsp release a new sipann? 

Sure!

joamatab, 2022-06-13:
That PR was closed a while ago 

A new PR based on current master would be great

SkandanC, 2022-06-17:
Hello @joamatab,

I was looking at the ring_single errors, and I modified the component to use Simphony's built in single bus ring resonator (built using sipann) component instead of approximating it with a straight waveguide and a half ring. 

I got much closer to the ideal values this time:

[image]

What do you think about this?

joamatab, 2022-06-17:
Looks good, also the add grating couplers function did not work after the last merge request

SkandanC, 2022-06-17:
Could you post a screenshot or a snippet of the error?

joamatab, 2022-06-17:
Sure, you can also just run the add_gc.py file

closed 2022-07-22 by joamatab