gdsfactory/gdsfactory #394: No electrical ports on heater cross_sections

thomasdorch, 2022-05-18:
In the heater CrossSections, it would be nice to have the heater ports added to the heater Section.
Like this:
> Section(width=1,  layer=LAYER.HEATER, port_names=port_names_electrical, port_types=port_types_electrical)

rather than:
> Section(width=1,  layer=LAYER.HEATER)

I tried doing this but pytest threw some errors regarding the port 'e1' already existing. Before I refactor things to make this work, is there any reason these CrossSections aren't set up like this already?

joamatab, 2022-05-18:
We can create a new crossSection with heater ports. The ones in the generic PDK Require a via stack so all the heater waveguides take the ports from the via stack

joamatab, 2022-05-20:
Hi Thomas,

I added electrical ports to the heater cross-sections as you suggested and released 5.7.1

thank you for your suggestion

closed 2022-05-20 by joamatab

thomasdorch, 2022-05-20:
Great, thank you!