gdsfactory/gdsfactory #396: Offsets generate route errors

thomasdorch, 2022-05-19:
get_bundle_waypoints tries to reverse the offset between ports (line 243, 244):

`    if start_angle in [90, 270]:
        offsets_start = [-_d for _d in offsets_start]`

When routing from a set of 270-degree ports to a set of 0-degree ports to their SW, the router throws a route error. Below is a minimum example of this behavour:

`    import gdsfactory as gf

    c = gf.Component("pads_bundle_steps")
    pt = c << gf.components.pad_array(
        gf.partial(gf.components.pad, size=(30, 30)),
        orientation=270,
        columns=3,
        spacing=(50, 0),
    )
    pb = c << gf.components.pad_array(orientation=0, columns=1, rows=3)
    pt.move((500, 500))

    routes = get_bundle_from_steps_electrical(
        pt.ports,
        pb.ports,
        end_straight_length=60,
        separation=30,
        steps=[{"dy": -50}, {"dx": -100}],
    )

    for route in routes:
        c.add(route.references)

    c.show()`

This can be fixed by removing the 270 from the list:
 `   if start_angle in [90]:
        offsets_start = [-_d for _d in offsets_start]`

joamatab, 2022-05-19:
Awesome, thank you Thomas

joamatab, 2022-05-19:
the issue with that change is that it breaks the bundle_u_indirect test

[image]

maybe we can add a flag to remove the offset start with 270 degrees angle,
what do you think?

joamatab, 2022-05-19:
I think we can change that line

it fixes the issue

[image]

joamatab, 2022-05-19:
it works for all the tests cases we have, im releasing 5.7.0 with all your new improvements, thank you Thomas!

joamatab, 2022-05-19:
just released 5.7.0

closed 2022-05-19 by joamatab