gdsfactory/gdsfactory #401: support routes with `None` orientation ports

joamatab, 2022-05-19:
How can we support bundle routes with ports that have orientation=`None` ?

```
if __name__ == "__main__":
    import gdsfactory as gf

    c = gf.Component("pads_bundle_steps")
    pt = c << gf.components.pad_array(
        gf.partial(gf.components.pad, size=(30, 30)),
        orientation=None,
        columns=3,
        spacing=(50, 0),
    )
    pb = c << gf.components.pad_array(orientation=None, columns=1, rows=3)
    pt.move((500, 500))

    routes = gf.routing.get_bundle_from_steps_electrical(
        pt.ports,
        pb.ports,
        end_straight_length=60,
        separation=30,
        steps=[{"dy": -50}, {"dx": -100}],
    )

    for route in routes:
        c.add(route.references)

    c.show()

```

tvt173, 2022-09-13:
do we support routes with `None` orientation ports? or neither bundles or routes at the moment? i would think it should just take an extra step of picking the most direct orientation to follow before routing

joamatab, 2022-09-13:
Not yet

PRs are most welcome:)

@SkandanC

SkandanC, 2022-09-13:
@joamatab 

I added a piece of code to add waypoints for None orientation ports as an "initial guess". The problem now is sorting the ports. The routing function fails when the port orientation is None, since the relative orientations of the input and output ports cannot be calculated.
[image]

But when I hard code in the port orientations just before sorting, I get this:

[image]

SkandanC, 2022-09-14:
Fixed it (this is with `None` orientation)
[image]

Another example:
```
import gdsfactory as gf

    c = gf.Component("get_route_from_steps_sample")
    w = gf.components.array(
        gf.partial(gf.components.straight, layer=(2, 0)),
        rows=3,
        columns=1,
        spacing=(0, 50),
    )

    left = c << w
    right = c << w
    right.move((200, 100))
    p1 = left.get_ports_list(orientation=None)
    for port in p1:
        port.orientation = None
    p2 = right.get_ports_list(orientation=None)
    for port in p2:
        port.orientation = None
    routes = get_bundle_from_steps(
        p1,
        p2,
        steps=[{"x": 150}, {"y": 100}],
    )

    for route in routes:
        c.add(route.references)

    c
    c.show()
```
[image]

@tvt173 @joamatab

joamatab, 2022-09-14:
Amazing! Thank you Skandan

joamatab, 2022-09-14:
fixed in https://github.com/gdsfactory/gdsfactory/pull/702

closed 2022-09-14 by joamatab