gdsfactory/gdsfactory #438: Use LayerSpec in Component.show()

Jan-David-Black, 2022-06-03:
**Is your feature request related to a problem? Please describe.**
I am not sure how I can permanently overwrite the layer that `c.show()` puts the port triangles. Currently the default is hard coded to `(1, 12)`.

**Describe the solution you'd like**
I believe it would make sense to change [this line](https://github.com/gdsfactory/gdsfactory/blob/4d76db32674c3edb4d16260e3177ee29ef9ce11d/gdsfactory/component.py#L916) to use a string LayerSpec, so that it can be dynamically set by a PDK

tvt173, 2022-06-03:
I'd also argue that `show_ports` should be `False` by default. i think this confuses a lot of first-time users... 

_"why are these shapes on my gds? i didn't draw them..."_

joamatab, 2022-06-06:
nice catch Jan-David! and thanks for the suggestion Troy

let me know what you think of
https://github.com/gdsfactory/gdsfactory/pull/441

- `show_ports=False` by defaul
- when `show_ports=True`, use layer `PORT` for drawing the port markers

closed 2022-06-06 by joamatab

tvt173, 2022-06-06:
@joamatab , my only criticism here is, aren't we also using that `PORT` layer for other (permanent) annotations, i.e. for LVS?

if so, i think it would be good to put these on a separate layer so as not to create confusion (i.e. `SHOW_PORT`?). otherwise, you could get these temporary, informational labels mixed up with those which actually get exported with your component

Jan-David-Black, 2022-06-06:
Putting it on `PORT` was actually also my first thought. But I also see your point Troy. I am not sure however how much of a problem it is, specially with `show_ports` being `False`by default.  If you have stuff in the `PORT` layer without `show_ports`you know it is put in permanently.
On the other hand one could also put it on `SHOW_PORT` and allow to coalesce `SHOW_PORT`and `PORT` into the same GDS layer, providing fine grain control to PDK owners to keep them separate or merged at will.

tvt173, 2022-06-06:
my other follow-up comment that I apparently forgot to make... 😆  

@joamatab, it would be really nice to have better documentation of the default/required PDK layers (maybe somewhere [here](https://gdsfactory.github.io/gdsfactory/notebooks/08_pdk.html)?), so we know which layers we need to add to our PDK and what they are used for without so much trial and error. Just a simple table of layer names and descriptions I think would be really helpful!

or if we do add a BaseModel of these required layers as discussed in the thread with @thomasdorch , you could just add descriptions for each and let sphinx auto-generate the docs, which may be easier to maintain!