gdsfactory/gdsfactory #439: Meep initialization time

thomasdorch, 2022-06-04:
Is it normal for 2D Meep simulations to be much slower to initialize than to actually run? I have been using the conda-forge Meep package with MPICH and write_sparameters_meep_mpi can take upwards of 5-10 minutes to initialize and perform subpixel averaging on a simulation domain of 20 x 10 at 30-40 resolution using 24 threads, but the simulation itself only takes maybe 30 seconds to a minute. 

If this is normal, is there anything we can do to speed up this part of the simulation?

tvt173, 2022-06-04:
@smartalecH

smartalecH, 2022-06-04:
24 threads or 24 processes?

If you're launching 1 process, but trying to use 24 threads, then it still builds the simulation domain in serial (no multi threading here due to the python callbacks). So yes, it could be rather slow depending on the geometry. 

If you are launching 24 processes (1 thread per process), however, it should initialize very quickly. The speedup is essentially linear for the initialization. Not true for the actual simulation, of course...

simbilod, 2022-06-04:
@smartalecH [the official way with mpirun](https://meep.readthedocs.io/en/latest/Parallel_Meep/#using-parallel-meep) spawns multiple processes right?

How it's done here: to allow a user to launch MPI simulations from _within_ a gdsfactory Python script, `write_sparameters_meep_mpi` pickles the gdsfactory component and simulation parameters, writes a temporary script that loads these settings and defines the MEEP simulation, and then runs this script with the `subprocess` call `mpirun -np {cores} python {temp_script_file}`.

I compared two simulations, one using `write_sparameters_meep_mpi` with 10 cores, and one where I directly called `mpirun -np 10 python temp_script_file.py`, and the timings were similar, so I don't think `Popen` is having unexpected behaviour.

smartalecH, 2022-06-05:
> @smartalecH [the official way with mpirun](https://meep.readthedocs.io/en/latest/Parallel_Meep/#using-parallel-meep) spawns multiple processes right?

Well, it depends... we added multithreading support (via OpenMP) to the codebase a ways back, so you can technically spawn multiple processes and threads *simultaneously* using `mpirun`/`mpiexec`. Just set `OMP_NUM_THREADS` to the thread count per process you want. We haven't done a good job documenting all the nuances here, unfortunately...

But I think I understand what you're really asking -- and the answer is yes. Your test looks good to me.

Just looking over the `gdsfactory` codebase, it looks like there is a "lazy parallel" option which runs multiple simulations in parallel using MPI. This would result in slow initializations, of course.

@thomasdorch if you are sure that you are truly running your simulation on 24 processes and still see significantly slow initializations, maybe pull together a MWE because it could indicate a bug. Maybe you can post some of the output of a run you are performing? How are you defining your geometry (material functions are very slow compared to geometry objects...)?

For the record, I regularly run large 3D simulations (15x15x10 @ resolution 50) on a 24-core machine and the initialization only takes a couple of minutes...

HelgeGehring, 2022-06-29:
How much points do your polygons have?
The subpixel averaging takes quite some time if you have polygons with a lot of points. For me it was far faster when I splitted everything up in polygons with a low number of points.

thomasdorch, 2022-06-29:
Thanks everyone for your help! I think the issue was the number of points in the polygons, as @HelgeGehring suggests.

I've been trying to put together a MWE to reproduce this issue using a straight waveguide but haven't been seeing the same issue with initialization time. Originally though, I was simulating an Euler bend, so the number of points was a lot larger in that simulation and that would certainly explain why I wasn't able to reproduce it with a straight waveguide..

HelgeGehring, 2022-06-29:
https://github.com/NanoComp/meep/issues/1271

joamatab, 2022-06-30:
Hi Thomas,

you can try to reduce the number of points of the bend to speed things up,

if you can share the code we can take a look

@HelgeGehring is looking into 2.5D simulations in meep

and @flaport  is looking into EME, which combined with the tidy3d mode solver that supports bend, would be a more elegant and faster way to solve large euler bends

thomasdorch, 2022-07-01:
@joamatab I've just been simulating bends for strip waveguides. I ran a few tests on the component below and extracted the `time for set_epsilon` directly from the Meep output for the following plots.

```
import gdsfactory as gf

bend = gf.components.bend_euler(radius=5)
write_sparameters_meep_mpi(
    component=bend,
    tpml=1,
    ymargin=1,
    zmargin_top=1,
    zmargin_bot=1,
    xmargin_right=1,
    port_margin=0.75,
    resolution=30,
)
```
Which looks like this with `npoints=300` and at `resolution=30`:
[image]

For different resolutions the time to set epsilon in a 2D simulation is linear with the number of vertices:
[image]

So I think my issue is the result too many vertices in the bends/too high of a resolution. The addition of 2.5D and/or EME will make rib waveguides a lot nicer to simulate so I'm very much looking forward to those!

HelgeGehring, 2022-07-01:
https://gdsfactory.github.io/gdsfactory/notebooks/plugins/meep/001_meep_sparameters.html#2.5D-FDTD

closed 2022-07-07 by thomasdorch