gdsfactory/gdsfactory #471: remove picwriter depedency

joamatab, 2022-06-21:
15% of gdsfactory module load time is on picwriter

[image]

quick solutions:

- move the import inside the functions

best solution:

- rewrite the picwriter based components natively into gdsfactory

joamatab, 2022-07-10:
the issue with the picwriter components is that it does not use the Path and CrossSection from gdsfactory, so it cannot be extruded with arbitrary cross-sections.

```
disk.py
spiral.py
dbr_tapered.py
coupler_full.py
coupler_adiabatic.py
grating_coupler_circular.py
cdc.py

```

Advantages of porting picwriter components:

- make gdsfactory module faster to load
- components can have arbitrary cross-sections

SkandanC, 2022-08-12:
@joamatab,

I tried moving the import inside the functions, but that doesn't seem to make a huge difference. I guess rewriting the picwriter-based components natively would be the better option. Would you like me to take a look into that?

Before moving import statements:
[image]

After moving import statements:
[image]

joamatab, 2022-08-13:
Hi Skandan,

Yes,
I think porting the few components that we use would be the best solution

joamatab, 2022-08-15:
https://github.com/gdsfactory/gdsfactory/pull/600

joamatab, 2022-08-16:
https://github.com/gdsfactory/gdsfactory/pull/601

We are half way done

Thank you Skandan

joamatab, 2022-08-19:
2 more to go :)

thank you Skandan!

SkandanC, 2022-08-19:
[image]
Down to 9% now :)

joamatab, 2022-08-20:
Nice, 

I think we can skip the circular grating coupler 

Once we implement the spiral we can remove picwriter

PR #631 (2022-08-23): remove picwriter dependency and lazy load scipy functions from init
remove picwriter dependency and lazy load scipy functions from init

closes #471 

Thanks to @SkandanC 

gdsfactory loads now in 1 second

[image]
closed 2022-08-23 by joamatab