gdsfactory/gdsfactory #483: adapt port markers when ports have a shear angle

joamatab, 2022-06-24:
How can we adapt the port markers when the port has a shear angle?

[image]

```
"""Port marker would be nice to understand the Shear angle.
"""

import numpy as np
import gdsfactory as gf

if __name__ == "__main__": 
    P = gf.path.straight(length=10)

    s = gf.Section(width=3, offset=0, layer=gf.LAYER.SLAB90, name="slab")
    X1 = gf.CrossSection(
        width=1,
        offset=0,
        layer=gf.LAYER.WG,
        name="core",
        port_names=("o1", "o2"),
        sections=[s],
    )
    s2 = gf.Section(width=2, offset=0, layer=gf.LAYER.SLAB90, name="slab")
    X2 = gf.CrossSection(
        width=0.5,
        offset=0,
        layer=gf.LAYER.WG,
        name="core",
        port_names=("o1", "o2"),
        sections=[s2],
    )
    t = gf.path.transition(X1, X2, width_type="linear")
    c = gf.path.extrude(P, t, shear_angle_start=10, shear_angle_end=45)
    c.show(show_ports=True)

```

tvt173, 2022-08-12:
hi @joamatab , to get the simple case, this should be fairly easy. one only needs to consider the shear_angle parameter when calling the function `get_pin_triangle_polygon_tip(port)` to get the two points of the triangle on the waveguide face. However, to get the more complicated case, where the port has a shear_angle AND a non-constant cross section, I believe you would need to annotate information about the endpoints of the face while extruding (i.e. `port.info['face'] = ...`), and use that to get those 2 points of the triangle. In all cases, the third point of the triangle need not be different from what it is now.

tvt173, 2022-08-12:
was able to get the more general case working. will send out a PR shortly

here's the result of your sample script above:
[image]

joamatab, 2022-08-12:
Thank you Troy

closed 2022-08-15 by joamatab