gdsfactory/gdsfactory #503: nicer docstrings

joamatab, 2022-06-30:
How can we render better docstrings?

See for example
https://gdsfactory.github.io/gdsfactory/components.html#mzi1x2-2x2

@flaport

flaport, 2022-06-30:
Do you mean docstrings or call signatures? For me the call signature looks a lot more scary than the docstringe...

I personally like autodoc_typehints='description' to relieve some of the pressure on the call signature.

flaport, 2022-06-30:
Also, if the type hint becomes too big it's worth aliasing your type hint for a shorter version.

joamatab, 2022-06-30:
nice, `autodoc_typehints='description' ` worked great!

how could we also show a better function name for partials?

```
add_pins=functools.partial(<function add_pins_siepic>
```

joamatab, 2022-06-30:
`autodoc_typehints='description'` merged

i also added a css to do one parameter per line

https://github.com/gdsfactory/gdsfactory/pull/510

closed 2022-08-05 by joamatab