gdsfactory/gdsfactory #505: Meep integration: cell size not correctly calculated

HelgeGehring, 2022-06-30:
The size of the Mode monitor is not correctly taken into account for bends:

```
from gdsfactory.components import bend_euler
from gdsfactory.simulation.gmeep import write_sparameters_meep
from gdsfactory.simulation.effective_permittivity import calculate_effective_permittivity

bend = bend_euler(radius=5)
material_name_to_meep = {
    'si': float(calculate_effective_permittivity(3.477 ** 2, 1.444 ** 2, 1.444 ** 2, .22, 1.55, 'te')[0] ** .5)}
write_sparameters_meep(bend, resolution=20, is_3d=False, animate=True, material_name_to_meep=material_name_to_meep)
```
[image]

joamatab, 2022-06-30:
yes, we could pass some padding, or you can use the bend90 function that we have in the docs

https://gdsfactory.github.io/gdsfactory/notebooks/plugins/meep/001_meep_sparameters.html#Port-symmetries

```
df = gm.write_sparameters_meep_1x1_bend90(c, run=False)
```

simbilod, 2022-06-30:
See PR #512

joamatab, 2022-06-30:
Hi Simon

i'd rather set the simulation manually, as we do now

the current solution:

- minimizes extra space
- lets you control the margin

```
write_sparameters_meep_mpi_1x1_bend90 = gf.partial(
    write_sparameters_meep_mpi,
    ymargin_bot=3,
    ymargin=0,
    xmargin_right=3,
    port_symmetries=port_symmetries.port_symmetries_1x1,
)
```
what do you think?

simbilod, 2022-06-30:
[Agreed](https://github.com/gdsfactory/gdsfactory/pull/512#issuecomment-1171498325)

joamatab, 2022-06-30:
another way to automatically fix these issues is to add `port_markers` 

[image]
[image]

HelgeGehring, 2022-06-30:
> another way to automatically fix these issues is to add `port_markers`
> 
> [image]

seems like the best approach!

simbilod, 2022-06-30:
This is a good opportunity to allow these port markers to hold height and mode information to allow multilayer and multimode simulations #502

joamatab, 2022-08-10:
see function to [add port markers in the docs](https://gdsfactory.github.io/gdsfactory/notebooks/plugins/meep/001_meep_sparameters.html#3D-rendering)

closed 2022-08-10 by joamatab