gdsfactory/gdsfactory #514: Forward https://gdsfactory.github.io/ to https://gdsfactory.github.io/gdsfactory

Jan-David-Black, 2022-06-30:
**Is your feature request related to a problem? Please describe.**
Somehow my browser always directs me to https://gdsfactory.github.io/. Would be nice If one was forwarded to https://gdsfactory.github.io/gdsfactory

HelgeGehring, 2022-07-03:
We would need a second repository with the name gdsfactory.github.io within the gdsfactory project for that with a page which just makes the redirection (https://www.w3docs.com/snippets/html/how-to-redirect-a-web-page-in-html.html)
```
<!DOCTYPE html>
<html>
  <head>
    <meta http-equiv="refresh" content="0; url='https://gdsfactory.github.io/gdsfactory'" />
  </head>
  <body>
    <p><a href="https://gdsfactory.github.io/gdsfactory">Gdsfactory Documentation</a>.</p>
  </body>
</html>
```
https://pages.github.com/
@joamatab

joamatab, 2022-07-03:
Yay! thank you @HelgeGehring for the fix

https://github.com/gdsfactory/gdsfactory.github.io

closed 2022-07-03 by joamatab