gdsfactory/gdsfactory #516: schematic capture

joamatab, 2022-07-01:
[image]

How can we make schematic capture of the circuits?

we currently support netlist flow

it would be great to support something like this for analog design flow

thomasdorch, 2022-07-01:
Maybe something like [SchemDraw](https://schemdraw.readthedocs.io/en/latest/index.html)?

joamatab, 2022-07-02:
Yes, ideally we can get

- automated completion of the circuit elements and ports for the active pdk
- hierarchical schematics 

The current YAML has some limitations 

@tvt173 
@flaport 
@proppy

joamatab, 2022-07-11:
https://xschem.sourceforge.io/stefan/xschem_man/what_is_xschem.html
https://github.com/StefanSchippers/xschem

Im thinking that with some graphviz visualization on networkx we could support schematic capture

proppy, 2022-07-11:
For python bases schematic capture, I looked a bit at https://github.com/devbisme/skidl but I think the current component model of gdsfactory might just be enough.

i.e a schematic is really just a graph of `Component` without size or position.

wdyt?

proppy, 2022-07-11:
PDKMaster also demonstrated python bases schematic capture and layout at FSiC22: https://gitlab.com/Chips4Makers/c4m-pdk-freepdk45/-/blob/master/design/inverter.ipynb

joamatab, 2022-07-12:
Yes, we can plot netlist from a component with networkX 

Ideally we can also allow users to reorder the graph, either with code or gui-like interactions adding bokeh or plotly callbacks 

@flaport

proppy, 2022-07-12:
@mithro also pointed us to https://github.com/nturley/netlistsvg before

joamatab, 2022-08-13:
https://github.com/StefanSchippers/xschem

Xschem is pretty nice for this

joamatab, 2022-10-11:
Here is a nice schematic driven flow, with auto-place and auto-route

https://www.youtube.com/channel/UCd2faHHQ_OAC0hSqBmIRKUQ/videos

with the YAML flow we have place and AutoRoute

joamatab, 2022-11-20:
-  https://qucat.org/tutorials/basics.html#Building-the-circuit-programmatically
- https://github.com/ntezak/cirq.git

joamatab, 2022-11-22:
https://neilturley.dev/netlistsvg/
https://github.com/nturley/netlistsvg

@tvt173

joamatab, 2022-11-27:
https://github.com/Vlsir/Hdl21Schematics

joamatab, 2022-11-29:
release 6.3.1 includes schematic driven flow

https://gdsfactory.github.io/gdsfactory/notebooks/20_schematic_driven_layout.html

thank you Troy!

closed 2022-11-29 by joamatab