gdsfactory/gdsfactory #521: remove phidl and gdspy dependencies from gdsfactory?

joamatab, 2022-07-05:
gdsfactory is built on top of phidl and gdspy

The main issues are:

- extra dependencies makes the package harder to install and sometimes hard to follow and document
- gdspy is no longer developed (only in maintenance mode). gdstk is supposed to be the successor of gdspy 
- gdspy requires conda install for windows. Now you can install gdsfactory through [mamba/conda](https://github.com/gdsfactory/gdsfactory/issues/226) but it would be nice to only depend on one package manager such as `pip`

We have some options

# 1. remove phidl and gdspy dependencies move to klayout API
- Klayout has faster booleans
- Klayout builds wheels for windows `pip install klayout`
- Klayout is already an optional dependency
- Klayout API is not super well documented

See @amccaugh work on the [phidl klayout backed](https://github.com/amccaugh/phidl/tree/klayout-backend)
@sebastian-goeldi 
@damienbonneau
@thomaslima

See [zeropdk](https://github.com/lightwave-lab/zeropdk)

# 2. remove phild and gdspy and move to gdstk

@heitzmann

# 3. Keep is as it is

- we benefit from all improvements in phidl as we are built on top of phidl, by the phidl community
- we can't easily export to OASIS

@thomasdorch 
@tvt173 
@flaport

thomasdorch, 2022-09-02:
I'm in favor of switching to the KLayout backend

thomaslima, 2022-09-02:
I obviously support the klayout option but disclaimer is I don't use gdsfactory yet as I focus on zeropdk. That said, next 0.27 version will support static type checking with mypy and docstring inspection from vscode, which should be helpful for you guys. See https://github.com/KLayout/klayout/pull/1125

alexsludds, 2022-09-03:
I would be in favor of migrating to the gdstk backend. My experience working with both gdspy and gdstk is that the APIs are similar, give the same layout results within ~1nm (so folks existing silicon is back compatible) and the compile times are much faster in gdstk.

tvt173, 2022-09-04:
I agree with @alexsludds . I think a migration to gdstk will be much more straightforward and there should be limited breaking changes because of the similarity between the two. i'm a big fan of klayout also and appreciative of the work from @thomaslima to make it easier to use the python API! but I'm not sure it makes sense to try and migrate gdsfactory over

thomaslima, 2022-09-09:
Hi @sebastian-goeldi I am curious about your comment about merging partial gds. I have to deal with this too and need to manually check for cell duplicates. Would you mind sharing an example and an expected behavior? I can try to implement a patch, or at least give a better error message.

BTW Here's my code for loading a specific cell from a gds file: https://github.com/lightwave-lab/zeropdk/blob/master/zeropdk/klayout_extend/layout.py

sebastian-goeldi, 2022-09-10:
Hi @thomaslima . I don't have an example directly at hand. I will try to get a minimal working example running when I get back from vacation ;). But I think the proper solution would be to run either a full `strm2gds` (I think that's the name of the routine) on the partial layouts or run the python equivalent with the LayoutDiff. With the diff object we can just throw errors/warnings if there's a difference between the two. If we don't care about names of cells (which I believe many do, including me), we could also just use conflict_resolution rename.

alexsludds, 2022-09-16:
Hey folks,
I posted this on a relevant thread on the phidl package, but I thought it important to put here too.
As part of converting gdsfactory away from gdspy I think it's important to move phidl away from gdspy to gdstk. I made a brief first pass at converting the phidl backend from gdspy to gdstk:
[phidl with gdstk backend](https://github.com/alexsludds/phidl)

It's not bug free, but it does generate some basic geometry. I need folks help with handling dependencies on gds/oas generation. 
In particular, if you run this example code (borrowed from the phidl examples):
[example code](https://github.com/alexsludds/phidl/blob/master/test_phidl_gdstk_backend.py)

You will see it tries to create three waveguides of different shapes. However, in the layout all waveguides have the same shape, equal to the last waveguide that was placed.

This is because of how I tried to implement the write_gds function:
[write_gds function](https://github.com/alexsludds/phidl/blob/c4a5abd8a485219fca940834474bcc1e66b74cee/phidl/device_layout.py#L1383)

I spent a few hours trying different ways to implement phidl's write_gds and have it be compatible with gdstk, but I can't get the dependency handling correct. I'm hoping this might be really obvious to someone else.

Best,
-Alex

joamatab, 2022-09-19:
thank you Alex, it looks good, I fixed some more tests

Another alternative is to create an intermediate format (such as shapely) and use klayout to read and write gds files

this is what Helge does in [gdshelpers](https://github.com/HelgeGehring/gdshelpers)

@HelgeGehring 
@jenshnielsen
@Dominik-Vogel

joamatab, 2022-10-20:
For example, the VLSIR project uses protocol buffers as an intermediate representation, and it has a rust package from converting to procol buffer to GDS

https://github.com/dan-fritchman/Layout21/issues/22

joamatab, 2022-11-08:
After the [gdstk segmentation issues](https://github.com/heitzmann/gdstk/issues/122)

Here is an MVP with [klayout backend](https://github.com/gdsfactory/gdsfactory/tree/klayout_backend)

@growly is also looking into using VLSIR as a backend

joamatab, 2022-11-10:
We were able to get rid of segmentation faults in gdstk by avoiding inheritance from gdstk.Polygon and gdstk.Label

gdsfactory 6.0.0 will be released soon using gdstk backend :tada: 

we also have an MVP of a `klayout_backend` on a branch that we will further investigate

closed 2022-11-10 by joamatab