gdsfactory/gdsfactory #553: docs issues

joamatab, 2022-07-29:
there are some issues in the docs

https://github.com/gdsfactory/gdsfactory/runs/7584210854?check_suite_focus=true

@flaport

flaport, 2022-07-29:
It seems like the pydata_sphinx_theme is trying to set an attribute that does not exist. It's an underscore attribute so it probably got refactored away in sphinx. I would try downgrading sphinx to an earlier version and see if that fixes the build.

joamatab, 2022-07-30:
Actually i realized that even the master docs test started failing 3 days ago

This [run succeeded 3 days ago](https://github.com/gdsfactory/gdsfactory/runs/7537947055?check_suite_focus=true)

And since then they have been failing [see](https://github.com/gdsfactory/gdsfactory/runs/7588910954?check_suite_focus=true)

The only difference I see is 
```
sphinx-markdown-tables==0.0.16
```

list of packages working
```
 alabaster-0.7.12 autodoc-pydantic-1.7.2 autotyping-21.12.0 babel-2.10.3 cfgv-3.3.1 coverage-6.4.2 distlib-0.3.5 doc8-0.11.2 docutils-0.17.1 filelock-3.7.1 flake8-4.0.1 flake8-bugbear-22.7.1 identify-2.5.2 imagesize-1.4.1 libcst-0.4.7 markdown-it-py-2.1.0 mccabe-0.6.1 mdit-py-plugins-0.3.0 mdurl-0.1.1 mypy-0.971 mypy-extensions-0.4.3 myst-parser-0.18.0 nbsphinx-0.8.9 nbval-0.9.6 nodeenv-1.7.0 pbr-5.9.0 platformdirs-2.5.2 pre-commit-2.20.0 pur-7.0.0 pycodestyle-2.8.0 pydata-sphinx-theme-0.8.1 pydocstyle-6.1.1 pyflakes-2.4.0 pytest-cov-3.0.0 recommonmark-0.7.1 restructuredtext-lint-1.4.0 snowballstemmer-2.2.0 sphinx-4.5.0 sphinx-autodoc-typehints-1.18.3 sphinx-book-theme-0.3.3 sphinx-click-4.3.0 sphinx-copybutton-0.5.0 sphinx-markdown-tables-0.0.16 sphinxcontrib-applehelp-1.0.2 sphinxcontrib-devhelp-1.0.2 sphinxcontrib-htmlhelp-2.0.0 sphinxcontrib-jsmath-1.0.1 sphinxcontrib-qthelp-1.0.3 sphinxcontrib-serializinghtml-1.1.5 stevedore-4.0.0 toml-0.10.2 tox-3.25.1 types-waitress-2.1.4 typing-inspect-0.7.1 virtualenv-20.16.1 xdoctest-1.0.1
```

list of packages failing
```
 alabaster-0.7.12 attrs-19.3.0 autodoc-pydantic-1.7.2 autotyping-21.12.0 babel-2.10.3 cfgv-3.3.1 coverage-6.4.2 distlib-0.3.5 doc8-1.0.0 docutils-0.19 filelock-3.7.1 flake8-4.0.1 flake8-bugbear-22.7.1 identify-2.5.2 imagesize-1.4.1 libcst-0.4.7 markdown-it-py-0.4.9 mccabe-0.6.1 mypy-0.971 mypy-extensions-0.4.3 myst-parser-0.8.2 nbsphinx-0.8.9 nbval-0.9.6 nodeenv-1.7.0 pbr-5.9.0 platformdirs-2.5.2 pre-commit-2.20.0 pur-7.0.0 pycodestyle-2.8.0 pydata-sphinx-theme-0.4.3 pydocstyle-6.1.1 pyflakes-2.4.0 pytest-cov-3.0.0 recommonmark-0.7.1 restructuredtext-lint-1.4.0 snowballstemmer-2.2.0 sphinx-5.1.1 sphinx-autodoc-typehints-1.19.0 sphinx-book-theme-0.0.39 sphinx-click-4.3.0 sphinx-copybutton-0.5.0 sphinx-markdown-tables-0.0.17 sphinxcontrib-applehelp-1.0.2 sphinxcontrib-devhelp-1.0.2 sphinxcontrib-htmlhelp-2.0.0 sphinxcontrib-jsmath-1.0.1 sphinxcontrib-qthelp-1.0.3 sphinxcontrib-serializinghtml-1.1.5 stevedore-4.0.0 toml-0.10.2 tox-3.25.1 types-waitress-2.1.4 typing-inspect-0.7.1 virtualenv-20.16.2 xdoctest-1.0.1

```

differences

```
docutils==0.17.1
pydata-sphinx-theme==0.8.1
sphinx-markdown-tables==0.0.16 
```

joamatab, 2022-07-30:
This is the fix

https://github.com/gdsfactory/gdsfactory/pull/554/commits/9e173804123851f3f67749bd21bf974a3bd2e8a7

closed 2022-07-30 by joamatab

0dminnimda, 2023-01-06:
> This is the fix
> 
> [9e17380](https://github.com/gdsfactory/gdsfactory/commit/9e173804123851f3f67749bd21bf974a3bd2e8a7)

I also found fixing `sphinx-book-theme==0.3.3` to solve this problem, thanks!