gdsfactory/gdsfactory #602: improve TCAD examples

joamatab, 2022-08-16:
How could we add better TCAD examples?

for example model a [PN junction](https://optics.ansys.com/hc/en-us/articles/360042328674-PN-depletion-phase-shifter)

@simbilod

simbilod, 2022-08-16:
The current DEVSIM code solves for carrier distribution in an arbitrary PN junction or PIN junction ridge waveguide as a function of DC voltage

We would need to export that carrier distribution (currently it exists on an unstructured tecplot mesh), and convert it to an index profile to input into one of the mode solvers

simbilod, 2022-08-16:
this might be a good time to start abstracting out geometry/mesh from the different solvers. Like have a "simulation domain" object that can call the different solvers on itself

joamatab, 2022-08-16:
perfect, looking forward to it

joamatab, 2022-08-18:
See the curves that would be nice to be able to get with DEVSIM

https://optics.ansys.com/hc/en-us/articles/360042328674-PN-depletion-phase-shifter

mdecea, 2022-09-09:
As far as I know, there is no way to import a spatial distribution of refractive index (or permittivity) into tidy3d. Do you guys know if it is possible? Haven't found it in the documentation. 
MPB should be able to do it.

simbilod, 2022-09-09:
Curved reproduced in #686

simbilod, 2022-09-09:
> As far as I know, there is no way to import a spatial distribution of refractive index (or permittivity) into tidy3d. Do you guys know if it is possible? Haven't found it in the documentation. MPB should be able to do it.

This was just implemented in this last PR :)

mdecea, 2022-09-09:
Awesome! Did not realize.

simbilod, 2022-09-09:
@mdecea 

See new `dn_dict` argument in tidy3D waveguide https://github.com/gdsfactory/gdsfactory/blob/master/gdsfactory/simulation/gtidy3d/modes.py

It takes in an unstructured mesh x, y, dn, and interpolates it onto the tidy3D index grid

You need to make sure the units of x, y that you pass fit with the mode solver coordinates

joamatab, 2022-09-09:
https://github.com/gdsfactory/gdsfactory/pull/686

we get agreement with other simulations

closed 2022-09-09 by joamatab

mdecea, 2022-09-12:
@simbilod 

I have noticed that in the `ramp_voltage` method you always start from V = 0 and ramp to the target voltage. If we are taking data at different voltages this is a bit inefficient (for example when getting effective index vs voltage) since we are getting the solution at the same voltage multiple times.

Do you know if we could just start ramping the voltage from the last voltage point we have already simulated?  I'm not familiar enough with devsim to know if it internally saves the last simulation but I would assume so?

simbilod, 2022-09-12:
Hi @mdecea , yes this was the case in my first PR. 

In the second PR I[ have changed `ramp_voltage` to take in Vinit and Vfinal,](https://github.com/gdsfactory/gdsfactory/blob/b96ae31c9de911fff0369c19129a7a6e7544e37d/gdsfactory/simulation/devsim/get_simulation_xsection.py#L399) so the user can more efficiently collect data. [This is what I do in the updated notebook](https://github.com/gdsfactory/gdsfactory/blob/master/docs/notebooks/plugins/devsim/01_pin_waveguide.ipynb)

When running DEVSIM, yes, it seems to keep device state throughout a Python session