gdsfactory/gdsfactory #610: more PCells

joamatab, 2022-08-18:
Add Pcells from papers or other PDKs to the generic PDK

- [ ] Mzi modulator
- [ ] photodetector 
- [ ] coherent tx
- [ ] coherent rx
- [ ] ring modulator 
- [ ] capacitor
- [ ] resistor
- [ ] skywater130 
- [ ] global foundries 180

simbilod, 2022-08-18:
what layerstack should these be built out of? the generic one here https://gdsfactory.github.io/gdsfactory/notebooks/08_pdk.html#layers ?

joamatab, 2022-08-19:
Yes, we can use the generic pdk layer stack and then we can re target any fab

Xies3, 2022-09-17:
Hi Joaquin, I'm currently working with Mehdi in the sky130 project. Is there any documentation of how to generate p cells? Thank you very much.

joamatab, 2022-09-18:
https://gdsfactory.github.io/gdsfactory/notebooks/03_cells_autoname_and_cache.html

closed 2023-09-12 by joamatab