gdsfactory/gdsfactory #611: freecad plugin

joamatab, 2022-08-18:
To model FEM heater efficiency we could add an FEM plugin to gdsfactory

https://wiki.freecad.org/FEM_Workbench

what do you think?

@basnijholt
@simbilod

simbilod, 2022-08-18:
For both FEM and TCAD, the first step would be to have suitable meshing of the components (3D), or components cross-sections (2D)

I know we already have Trimesh for visualization, but having the option for adaptive meshing would be very helpful. Gmsh (used in the Freecad pipeline linked above) does this, see http://web4.cs.ucl.ac.uk/research/vis/toast/demo_meshgen3/demo_meshgen3.html

Once we have a mesh in one format we can easily convert to any other format with [meshio](https://github.com/nschloe/meshio)

Then any finite element solver would be good. Examples:
- DEVSIM (can implement arbitrary models, not just semiconductor)
- Freecad (above)
- [MOOSE](https://github.com/idaholab/moose)
- [FeNics](https://fenicsproject.org/)
etc.

joamatab, 2022-09-29:
fixed by #739

closed 2022-09-29 by joamatab