gdsfactory/gdsfactory #612: cloud storage for GDS and simulation results

joamatab, 2022-08-18:
We have been storing GDS and simulation results on a separate git repo, the issue is that the repo is currently 3.7GB

https://github.com/gdsfactory/gdslib/tree/data

How about creating a Google Cloud storage (GCS) bucket and mount it?

@proppy
@HelgeGehring

joamatab, 2022-08-20:
Another option is using the GitHub artifacts 

@flaport

flaport, 2022-08-20:
I use [DVC](https://dvc.org/) for this.

Just make a public GCS bucket and track large files in your repository with DVC. It's pretty convenient.

flaport, 2022-08-20:
You can even download your files on the fly as you need them

joamatab, 2022-08-21:
That's a great idea,
the option of mounting a cloud bucket seems to work for mac and linux only
https://cloud.google.com/storage/docs/gcs-fuse

what do you think of using dvc with a public bucket? See this ubcpdk branch
https://github.com/gdsfactory/ubc/tree/dvcubcpdk

flaport, 2022-08-21:
> what do you think of using dvc with a public bucket? See this ubcpdk branch https://github.com/gdsfactory/ubc/tree/dvcubcpdk

That's exactly how I would do it 🙂

joamatab, 2022-08-21:
Awesome 

For large data such as measurements we have been mounting the bucket, so we don't have to download everything 

How can we do that with dvc?

joamatab, 2022-08-23:
closes

https://github.com/gdsfactory/gdsfactory/pull/626

closed 2022-08-23 by joamatab