gdsfactory/gdsfactory #621: fast fully web based layout viewer

joamatab, 2022-08-19:
How can we enable big layouts rendering in jupyter notebooks?

we could use the klayout python API and use some thing like folium

- https://python-visualization.github.io/folium/quickstart.html#GeoJSON/TopoJSON-Overlays
- https://github.com/klayoutmatthias/canvas2canvas

@flaport 
@proppy
@tvt173

proppy, 2022-08-30:
I think that https://github.com/KLayout/klayout/issues/1082#issuecomment-1197720698 is the way to go.

Did you see the notebook example?

proppy, 2022-08-30:
I think ideally we would just implement some of the `_repr` methods so that `Component` just render magically when evaluated at the end of cell.

joamatab, 2022-11-28:
Where can we find the code for SiliWiz?

https://www.youtube.com/watch?v=V9xCa4RNfCM

@SkandanC 
@mithro 
@mattvenn

mattvenn, 2022-11-29:
It's not public at the moment

joamatab, 2022-11-30:
I see, let us know when we can use it,

also the 3D rendering on the browser looks great
https://www.youtube.com/watch?v=GlvlZzjyUGI&t=294s

joamatab, 2022-12-11:
klayout 0.28 is out

https://pypi.org/project/klayout/

https://github.com/klayoutmatthias/canvas2canvas

proppy, 2022-12-11:
@joamatab getting this issue when trying to use the new klayout 0.28:
https://github.com/KLayout/klayout/issues/1146#issuecomment-1345445086

Are you able to load a layout on your end?

joamatab, 2023-01-24:
See MVP here

https://github.com/gdsfactory/kweb/issues/3

joamatab, 2023-01-28:
TODO:

- [ ] Document how to deploy gdsfactory with kweb
- [ ] make an ipywidget for notebooks using klayout

sebastian-goeldi, 2023-02-03:
https://stackoverflow.com/questions/74070505/how-to-run-fastapi-application-inside-jupyter

joamatab, 2023-02-03:
that worked!

```
import asyncio
import uvicorn
from kweb.main import app

if __name__ == "__main__":
    config = uvicorn.Config(app)
    server = uvicorn.Server(config)
    loop = asyncio.get_event_loop()
    loop.create_task(server.serve())
```

SkandanC, 2023-02-03:
might be too late now but https://github.com/wokwi/siliwiz it's open source.

joamatab, 2023-02-03:
That looks great!

we have a jupyter demo

[image]

joamatab, 2023-08-03:
we have 

- https://github.com/gdsfactory/kweb
- gdsfactory webapp in https://github.com/gdsfactory/gplugins/

closed 2023-08-03 by joamatab