gdsfactory/gdsfactory #627: gdspy dependency is failing in python3.7 and python3.8 environments

saicharan0112, 2022-08-23:
**Describe the bug**
There is an installation error for gdspy package. This is how it looks like when you install it - 
[image]

**To Reproduce**
I created a venv for both python3.7 and 3.8. Let me know you are able to reproduce the error using this method.

**Expected behavior**
Clean installation

**Suggested fix**
No idea

saicharan0112, 2022-08-23:
Not sure if you have any idea on this fix. I don't see any development from past 2 months on the gdspy github repo.

saicharan0112, 2022-08-23:
@joamatab Not sure if you have any idea on gdspy codebase. I don't see any development (code) on the gdspy github repo.

joamatab, 2022-08-23:
Hi Sai,

to reproduce the error you should write all the commands that you used, a screenshot is not enough :)

my guess is that you did not follow the install instructions for [gdsfactory](https://gdsfactory.github.io/gdsfactory/notebooks/_2_install.html)

How did you install gdsfactory?
Did you follow the gdsfactory instructions and install it with mamba?

I also recommend that gdspy install issues should be addressed in the gdspy repo
https://github.com/heitzmann/gdspy

closed 2022-08-23 by joamatab

saicharan0112, 2022-08-24:
I added the screenshot under "Describe the bug" section cuz that is the only message I got. 

I installed gdsfactory via pip which is one of the dependencies in OpenFASoC. Whenever I start the installation, I get error in the gdspy installation section. I explicitly tried to install gdspy to confirm the error. 
These are the commands I used to insta gdspy - 
`python3.7 -m venv "my_env" &&
source ~/my_env/bin/activate &&
pip3 install gdspy`

Sure, I will open there as well.

tvt173, 2022-08-24:
@saicharan0112 please try installing pip<22. I believe this is an error with the latest pip

tvt173, 2022-08-24:
also are you on windows or linux? make sure to always install with conda if on windows

saicharan0112, 2022-08-24:
Sure, let me give it a try. I am using Linux CentOS7 BTW.

saicharan0112, 2022-08-24:
Thanks alot, @tvt173 . That downgrading version trick removed this issue. But I got a few other error messages. Can you please take a look at this log and help me understand how to handle this?
[gdsfactory_error.log](https://github.com/gdsfactory/gdsfactory/files/9416977/gdsfactory_error.log)

saicharan0112, 2022-08-24:
@tvt173 I think I fixed this. It is something to do with the python parallel installation. I didn't had required packages for the python3.7 to install properly. After that error, I followed the steps shown here (https://stackoverflow.com/questions/27022373/python3-importerror-no-module-named-ctypes-when-using-value-from-module-mul) and I got a clean installation of the gdsfactory (and gdspy) packages. 
I don't think this is something to do with the pip version cuz I am able to install it with pip 22.2.2 version.

But thanks for your help @tvt173 . If I didn't get that error shown in the log file, I would have overseen this library dependency issues.