gdsfactory/gdsfactory #646: Writing Grating with rectangle & array

zhixwang, 2022-08-30:
Hi, I'm trying to write a simple grating, consisting of a long row of straight bars, using gf array and rectangle:

```
bar1 = gf.components.rectangle(size=[lw1, bar_length], layer=1, centered=True)
c = gf.components.array(bar1
, columns=100000, rows=1, spacing=(pitch1, 0.0))

c.name = "Array1"
c.write_gds("demo.gds")
```

However, I encountered an error:
> Exception has occurred: error
> 'h' format requires -32768 <= number <= 32767

Can't I write an array with more than 32678 grating bars? How should I write a grating properly with this library?

Thanks!

Best,
Jason

tvt173, 2022-08-30:
Apparently it's a limitation of the GDS format:
https://www.artwork.com/gdsii/qis/faq/qis_limits.htm#:~:text=5.,cell%20array%20reference%20is%2032767.

You can certainly draw the raw polygons though... make a 3D numpy array (a list of polygons), then add with `c.add_poly(polys, layer=layer)`

zhixwang, 2022-08-30:
Thanks for the reply. Isn't there any built-in API for creating a normal grating directly?

I see a lot of similar but more complicated built-in component types, like AWG, grating coupler, ...

joamatab, 2022-08-30:
yes, you can use the grating from the generic pdk

```
import gdsfactory as gf

c = gf.components.grating_coupler_rectangular

```

https://gdsfactory.github.io/gdsfactory/components.html#grating-coupler-rectangular

closed 2022-08-30 by joamatab