gdsfactory/gdsfactory #648: rotate() got an unexpected keyword argument 'center'

zhixwang, 2022-08-30:
**Describe the bug**
rotate() got an unexpected keyword argument 'center'

**To Reproduce**
`bar = gf.components.rectangle(size=[bar_length, lw1], centered=True).rotate(ang1, center = [0,0])`

**Expected behavior**
In the Documentation, rotate() supports center as an argument.

joamatab, 2022-08-30:
Hi,

Where in the documentation?

https://gdsfactory.github.io/gdsfactory/_autosummary/gdsfactory.Component.html#gdsfactory.Component

zhixwang, 2022-08-30:
https://gdsfactory.github.io/gdsfactory/notebooks/02_movement.html#

```
c = gf.Component("rectangles")
r = gf.components.rectangle(size=(10, 5), layer=(2, 0))
rect1 = c << r
rect2 = c << r

rect1.rotate(45)  # Rotate the first straight by 45 degrees around (0,0)
rect2.rotate(
    -30, center=[1, 1]
)  # Rotate the second straight by -30 degrees around (1,1)
c
```

joamatab, 2022-08-30:
`rect` is not a `Component` but a `Component Reference`

`ComponentReference.rotate()` has a `center` method

closed 2022-08-30 by joamatab