gdsfactory/gdsfactory #653: replace sipann by sax neural network model

joamatab, 2022-08-31:
Sipann depends on tensorflow (500MB download) and does not have code or documentation to build new PDK models (for example in Silicon nitride)

It would be great to port SiPANN like models to JAX and use SAX as a circuit simulator 

This way instead of using tensorflow/sipann/simphony we could do circuit simulations in SAX

As far as I know there are two ways to do it:

1. mode solver (faster at but only computes single frequency). We could use tidy3d Mode solver
2. FDTD (slower but solves different frequencies). We could use meep

Floris, what do you think of using tidy3d mode solver versus meep as simulator?

@flaport

flaport, 2022-09-01:
To create sipann-like models you need lots of simulations as training data, so if possible I'd use mode solvers and EME approaches over FDTD. Ideally the models will be able to interpolate between wavelengths if you sample enough.

That said, some structures need FDTD to be accurately modeled. It remains a decision of the designer on how something should be most accurately modeled.

I think if we want to do something similar we should start with easy components which can be acccurately modeled with mode solvers. Once we get more comfortable making SiPANN-like models we should experiment with 2D FDTD and eventually 3D FDTD.

joamatab, 2022-09-01:
sounds like a good plan.

so we can start building Sparameter models for waveguides and couplers

the idea would be to return a Sax model for Silicon and Silicon Nitride, both for Cband (1.55um) and Oband (1.30um). in my experience Silicon index is very similar between fabs, but as SiN can change a lot it would be nice to document how to build your own models

Models:

- waveguide
- coupler (evanescent coupler)
- ring coupler

```
def waveguide(width:float=0.5, thickness:float=0.22, thickness_slab:float0, length:float=10)->sax.SDict:
     return fit_waveguide_using_jax_neural_network

```

@SkandanC

SkandanC, 2022-09-01:
So the idea is we port SiPANN to JAX (which I am working on atm), and use that in the SAX/SiPANN plugin to return SAX models?

joamatab, 2022-09-02:
Yes, and we can benchmark the sparameters that the model predicts with FDTD simulations and measurements

joamatab, 2023-02-14:
Interesting package with some more models

https://github.com/alexsludds/photonflux/blob/main/photonflux/component_models.py

it may be great to fit models using simulations and measurements

joamatab, 2023-09-12:
should be reopened in https://gdsfactory.github.io/gplugins/index.html

closed 2023-09-12 by joamatab